STMicroelectronics/STM32CubeF7 #3: EXTI-HAL (i.e. not GPIO-HAL using EXTI)

FerdinandEn, 2019-08-20:
Example Code with error
```
void myEXTIinit(void)
{
  EXTI_ConfigTypeDef myExtiConfig;
  
  HAL_EXTI_RegisterCallback(&hexti13, HAL_EXTI_COMMON_CB_ID , 
                              &myEXTI13IRQHandler);
  
  myExtiConfig.Line = EXTI_LINE_13;
  myExtiConfig.Mode = EXTI_MODE_INTERRUPT;
  myExtiConfig.Trigger = EXTI_TRIGGER_RISING;
  HAL_EXTI_SetConfigLine(&hexti13, &myExtiConfig);
}
```
HAL_EXTI_SetConfigLine() produces a Hard-Fault

### Reason:

```
HAL_StatusTypeDef HAL_EXTI_SetConfigLine(EXTI_HandleTypeDef *hexti, EXTI_ConfigTypeDef *pExtiConfig)
{
  uint32_t regval; // temporary variable to calculate peripheral address is a integer type not a pointer

// ...
// EXTI_MODE_INTERRUPT = 1
regval = (uint32_t)EXTI_BASE;
regval += pExtiConfig->Mode; // >> leads to odd number
*(__IO uint32_t *) regval |= pExtiConfig->Line; // illegal access for peripherals and wrong address for mode

//...
}
```

- Same problem for trigger rising or falling edge -> additional wrong base address for RTSR, FTSR

- Register EXTI-IMR is cleared but not set again by  HAL_EXTI_SetConfigLine() >> no interrupt

Please see also 
Case#: 00084085
Subject: STM32F7 EXTI-HAL (i.e. not GPIO-HAL using EXTI) STM32F7 Cube 1.15

Used Toolchain: Keil Compiler 5 or 6, CubeMX 5.3.0, HAL 1.15

CCASTM, 2019-08-21:
Hello Ferdinand,
Thank you for your problem report, with a good description. 
I have forwarded it to the development team and we will give you a feedback very soon.

I can see a Case# reference in your comment. Did you also submitted a support request to the On-Line-Support  interface ?
Regards
Christophe

FerdinandEn, 2019-08-21:
Hello Christophe,
I also reported the request via the support interface. However, the response times are very long. That's why I tried it this way.
Regards
Ferdinand

CCASTM, 2019-08-27:
Hello Ferdinand,
The development team has been solicited. I will trig them again 
Regards
Christophe

ALABSTM, 2019-09-03:
Hi Ferdinand,

Glad to read your message and thank you once again for pointing out this issue.

Actually, we already spotted the problem and fixed it. However, the fix has not been published yet. It is scheduled for next release (STM32CubeF7 v1.16.0).

In the meanwhile, please find attached this patch [STM32F7xx_HAL_Driver.zip](https://github.com/STMicroelectronics/STM32CubeF7/files/3570207/STM32F7xx_HAL_Driver.zip) implementing the fix. Once unzipped, please replace your current HAL EXTI .h and .c files with the ones within.

We hope this helps you solve the problem. Do not hesitate in case something still does not work properly.

With regards,

FerdinandEn, 2019-09-05:
Hello ALABSTM,

I tested the new code. Unfortunately it still doesn't work. As far as I can see, the EXTI registers are set correctly. The error now occurs in the register SYSCFG->EXTICR4. The pin I used is PC13. The corresponding entry EXTI13 in the EXTICR4 register would have to be set to 2. The used code changes the entries of EXTI13, EXTI14 and EXTI15. EXTI13 also has the wrong value 0x0D. After correction by hand in the debugger to 2 the driver works. The problematic section is as follows.

   ```
 /* Configure gpio port selection in case of gpio exti line */
    if ((pExtiConfig->Line & EXTI_GPIO) == EXTI_GPIO)
    {
      assert_param(IS_EXTI_GPIO_PORT(pExtiConfig->GPIOSel));
      assert_param(IS_EXTI_GPIO_PIN(linepos));

      regval = SYSCFG->EXTICR[linepos >> 2u];
      regval &= ~(SYSCFG_EXTICR1_EXTI0 << (SYSCFG_EXTICR1_EXTI1_Pos * (linepos & 0x03u)));
      regval |= (pExtiConfig->GPIOSel << (SYSCFG_EXTICR1_EXTI1_Pos * (linepos & 0x03u)));
      SYSCFG->EXTICR[linepos >> 2u] = regval;
    }
  }
```
Regards
Ferdinand

FerdinandEn, 2019-09-05:
Hello ALABSTM,
I found the problem. In the new version the datatype EXTI_ConfigTypeDef was extended by the value GPIOSel. I didn't set it. After that everything works fine. Please excuse the wrong message.

Regards
Ferdinand

ALABSTM, 2019-09-05:
Hi Ferdinand,

Glad to hear the problem is solved. I think then this issue could be considered as closed.

With regards,

closed 2019-09-05 by FerdinandEn

ALABSTM, 2019-09-05:
ST Internal Reference: 40803