STMicroelectronics/STM32CubeF7 #4: Inconsistent GPIO macro names

duskwuff, 2019-09-29:
I've encountered a couple of weird inconsistencies in names used for GPIO macros:

* There are no `_Pos` or `_Msk` macros for `OTYPER`, and the macros that do exist have an unexpected underscore between `OT` and the bit number. (For example, there is no `GPIO_OTYPER_OT0_Pos`.) Similar issues exist for `IDR`, `ODR`, and `BSRR`.

* The `OSPEEDR` register is misspelled in some places as `OSPEEDER`, e.g. `GPIO_OSPEEDER_OSPEEDR0_Pos`.

ALABSTM, 2019-09-30:
Hi Dusk,

The issue you pointed out has been communicated to our development teams. It will be investigated soon. We will be back to you as soon as we get an answer. Thank you for your patience.

With regards,

ALABSTM, 2019-11-04:
Hi Dusk,

Our development teams accepted your request. Definitions using _Pos and _Msk macros will be available in a future release.

Misspelled bit names GPIO_OSPEEDER_OSPEEDRx shall also be corrected.

With regards,

ALABSTM, 2019-11-04:
ST Internal Reference: 75108

ALABSTM, 2019-11-04:
ST Internal Reference: 75110

ALABSTM, 2020-02-28:
Hi Dusk,

I hope you are doing well. This issue has been fixed in the STM32CubeF7 **v1.16.0** firmware package which will be published soon (already available on our website "st.com").

I think we can close this issue now. Thank you again for your contribution.

With regards,

closed 2020-02-28 by ALABSTM