STMicroelectronics/STM32CubeF7 #5: Floating-point ABI for Audio Addons libraries

smartel99, 2019-10-10:
**Describe the set-up**
 * STM32F746G-DISCO
 * stm32CubeIDE

**Additional context**
The addon libraries of the STM32_Audio Middlewares are only available compiled using `-mfloat-abi=hard`, causing compatibility issues with software using `-mfloat-abi=soft` and `-mfloat-abi=softfp`.

Solution would be to provide libraries compiled with mixed HW/SW floating-point values

ALABSTM, 2019-10-16:
Hi Samuel,

Thank you for this message. Your request has been sent to our development teams. We will be back to you as soon as we get an answer. Thank you for your patience.

With regards,

ALABSTM, 2019-11-05:
Hi Samuel,

Our development teams confirmed your request will be addressed. Actually, the point you raised has also been discussed internally since a while, before we receive your request. A new compiled library for CM4 core with option `-mfloat-abi=softfp` will be made available in the frame of a future release.

We cannot share a date for the moment. Hopefully, this should be done soon. So stay tuned and thank you once more for your contribution and for your patience.

With regards,

ALABSTM, 2020-01-06:
Hi Samuel,

I hope you are doing well. The new compiled library will be available in the frame of version 1.16.0 of the STM32CubeF7 firmware package, which will be released soon, hopefully.

We do apologize for the delay. Thank you once more for your patience.

With regards,

smartel99, 2020-01-13:
Awesome, thank you!

ALABSTM, 2020-03-04:
Hi Samuel,

Firmware package STM32CubeF7 **v1.16.0** is finally available on GitHub. Inside the [STM32_Audio](https://github.com/STMicroelectronics/STM32CubeF7/tree/79acbf8ec060d3ec751f2eaba6ee050269995357/Middlewares/ST/STM32_Audio/Addons/PDM) folder (under the Middlewares), you will find a "readme.txt" file with a link within, which will redirect you to the "st.com" website.

You will be asked to register or to login (if not already logged). Then, your download should automatically start. This procedure is due to the fact that the "Audio PDM" component is subject to some particular license terms.

I hope this addresses your request. I think this issue can be closed now.

With regards,

closed 2020-03-04 by ALABSTM