STMicroelectronics/STM32CubeF7 #6: QSPI locks up at STM32F746G-DISCO board

desowin, 2019-11-22:
**Describe the set-up**
 * STM32F746G-DISCO
 * STM32746G-Discovery BSP V2.0.2 with STM32F7xx HAL Drivers V1.2.7
 * gcc 6.3.0 from TouchGFX 4.12.3 toolchain

**Describe the bug**
When repeadetly calling BSP_QSPI_Read() the QSPI locks up with BUSY flag set. The only way to get out of the issue is to reset using BSP_QSPI_DeInit() followed by BSP_QSPI_Init().

The problem only happens when optimization level `-O2` is active. It does not happen with `-O1`. 

**Additional context**
This bug is basically the same as https://github.com/ARMmbed/mbed-os/issues/10049.
The fix https://github.com/ARMmbed/mbed-os/commit/5c872a4c0669f9d2ef70a9401bbaa469b9881f57 when applied on the HAL library, solves the issue completely.

ALABSTM, 2019-11-26:
Hi Desowin,

This issue has been forwarded to our development teams. We will be back to you as soon as they provide us with feedback. Thank you for your contribution.

With regards,

ALABSTM, 2019-11-26:
Hi Desowin,

Our development teams confirmed the issue. A fix has been implemented. It shall be available in the next STM32CubeF7 release (v1.16.0).

We cannot share a date for the moment. Hopefully, it should be soon enough. Thank you for your patience and thank you once more for your contribution.

With regards,

ALABSTM, 2019-11-26:
ST Internal Reference: 64387

ALABSTM, 2020-04-13:
Hi @desowin,

I hope you are doing well. This issue has been fixed and the fix is now available in the STM32CubeF7 **v1.16.0** firmware package published some time ago. This issue can be closed now.

Thank you again for your contribution. Take care and stay safe.

With regards,

closed 2020-04-13 by ALABSTM