STMicroelectronics/STM32CubeF7 #7: USB_DriveVbus() doxygen documentation is incorrect

desowin, 2019-11-28:
**Describe the set-up**
 * STM32F746G-DISCO
 * gcc 6.3.0 from TouchGFX 4.12.3 toolchain

**Describe the bug**
USB_DriveVbus() doxygen comment is
```
/**
  * @brief  USB_DriveVbus : activate or de-activate vbus
  * @param  state  VBUS state
  *          This parameter can be one of these values:
  *           0 : VBUS Active
  *           1 : VBUS Inactive
  * @retval HAL status
*/
```
However the `state` is directly mapped to the `USB_OTG_HPRT_PPWR` bit that RM0385 Rev 8 describes as:
```
Bit 12 PPWR: Port power
The application uses this field to control power to this port, and the core clears this bit on an
overcurrent condition.
0: Power off
1: Power on
```

**Additional context**
This seems to be present in all STM32 series library.

desowin, 2019-11-28:
The same applies to `USBH_LL_DriverVBUS()` in USB Host examples.

desowin, 2019-11-28:
Another issue, is why the USB Host examples do not use `USB_DriveVbus()` in their `USBH_LL_DriverVBUS()` implementation when `USE_USB_HS` is defined?

ALABSTM, 2019-11-28:
Hi Tomasz,

Your remark will be transmitted to our development teams to check its relevance. You will be updated as soon as they provide us with an answer.

Regarding your second question, it is good practice to submit it in a separate issue. This would ease tracking and answering the different issues you would open. Hence, would you mind reporting it in another issue? Thank you for your comprehension.

With regards,

ALABSTM, 2019-11-29:
Hi Tomasz,

Please allow me to introduce @bouattay who will answer your question.

With regards,

ALABSTM, 2020-06-08:
Hi @desowin,

I hope you are doing well. Your remark is relevant. The description of the "state" parameter is incorrect. Our development teams will be notified to fix this.

Thank you for having reported this point. Take care and stay safe.

With regards,

ALABSTM, 2020-06-08:
ST Internal Reference: 87763

ALABSTM, 2021-03-08:
Hi @desowin,

Comments updated as you can see below. Thank you again for your contribution.

With regards, 

https://github.com/STMicroelectronics/STM32CubeF7/blob/c7c5ec99c7482ea8bcdbf0a869c930af4547088f/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_ll_usb.c#L1586-L1589

closed 2021-03-08 by ALABSTM