STMicroelectronics/STM32CubeF7 #8: USB Host does not work with DCache and DMA enabled

desowin, 2019-12-10:
**Describe the set-up**
 * STM32F746G-DISCO
 * STM32F7xx HAL Drivers V1.2.7
 * STM32 USB Host Library V3.3.2
 * gcc 6.3.0 from TouchGFX 4.12.3 toolchain
 * Modified USB Host configuration USB_Host/MSC_RTOS/Src/usbh_conf.c to contain `hhcd.Init.dma_enable = 1;` instead of `hhcd.Init.dma_enable = 0;`

**Additional context**
In my project I have FreeRTOS, STM32 USB Host Library and FatFs. The FatFs driver and Host Library configuration files are taken from MSC_RTOS example. The example does not use DMA and I wanted to enable it. I thought I could just change the flag `dma_enable` to `1` but it doesn't work unless I disable DCache.

I have captured the raw USB traffic using OpenVizsla USB analyzer. It is apparent that the SETUP transactions do not contain valid data (zeros only; however the CRC is correctly calculated for all zeros) when the configuration is changed to contain `hhcd.Init.dma_enable = 1;` and the main() contains `SCB_EnableDCache();`.

For example, the very first SETUP transaction sent to device when DMA is not used is:
```
USB Packet (2d 00 10)
    PID: SETUP (0x2d)
    [Source: host]
    [Destination: 0.0]
    .... .... .000 0000 = Address: 0
    .... .000 0... .... = Endpoint: 0
    0001 0... .... .... = CRC5: 0x02 [correct]
    [CRC5 Status: Good]
USB Packet (c3 80 06 00 01 00 00 08 00 eb 94)
    PID: DATA0 (0xc3)
    Data: 8006000100000800
    CRC: 0x94eb [correct]
    [CRC Status: Good]
USB Packet (d2)
    PID: ACK (0xd2)

```
to such request the device replies correctly wih the 8 bytes of the Device Descriptor, and the rest of enumeration and the actual FatFs driver works fine.

```
USB Packet (69 00 10)
    PID: IN (0x69)
    [Source: host]
    [Destination: 0.0]
    .... .... .000 0000 = Address: 0
    .... .000 0... .... = Endpoint: 0
    0001 0... .... .... = CRC5: 0x02 [correct]
    [CRC5 Status: Good]
USB Packet (4b 12 01 10 02 00 00 00 40 55 41)
    PID: DATA1 (0x4b)
    Data: 1201100200000040
    CRC: 0x4155 [correct]
    [CRC Status: Good]
USB Packet (d2)
    PID: ACK (0xd2)
```

However when the `SCB_EnableDCache();` is called during initialization and DMA is enabled in usbh_conf.c, the very first SETUP transaction sent to device is malformed:
```
USB Packet (2d 00 10)
    PID: SETUP (0x2d)
    [Source: host]
    [Destination: 0.0]
    .... .... .000 0000 = Address: 0
    .... .000 0... .... = Endpoint: 0
    0001 0... .... .... = CRC5: 0x02 [correct]
    [CRC5 Status: Good]
USB Packet (c3 00 00 00 00 00 00 00 00 bf f4)
    PID: DATA0 (0xc3)
    Data: 0000000000000000
    CRC: 0xf4bf [correct]
    [CRC Status: Good]
USB Packet (d2)
    PID: ACK (0xd2)
```

and the device (obviously) STALLs when host tries to read data:
```
USB Packet (69 00 10)
    PID: IN (0x69)
    [Source: host]
    [Destination: 0.0]
    .... .... .000 0000 = Address: 0
    .... .000 0... .... = Endpoint: 0
    0001 0... .... .... = CRC5: 0x02 [correct]
    [CRC5 Status: Good]
USB Packet (1e)
    PID: STALL (0x1e)
```

desowin, 2019-12-10:
The workaround is to either:
- disable DCache using `SCB_DisableDCache();` or by not enabling it in the first place
- not use DMA in USB Host

desowin, 2019-12-10:
Working, however not the best solution is to:
- Enable DMA by changing `hhcd.Init.dma_enable = 0;` to `hhcd.Init.dma_enable = 1;`
- Force Write-Through before enabling DCache: `SCB->CACR |= SCB_CACR_FORCEWT_Msk;`
- Invalidate DCache inside `HAL_HCD_HC_NotifyURBChange_Callback()`: `SCB_InvalidateDCache();`

That way, the changes are solely user code, in the `usbh_conf.c` and the place where `SCB_EnableDCache();` is called.

A better solution would be to ensure cache coherency at the STM32 Library level.

ALABSTM, 2019-12-12:
Hi Desowin,

Welcome back! Thank you for this detailed description of the behavior you observed. Thank you also for the suggestions you made.

Please let me introduce @bouattay once again. He will take in charge analyzing this issue and the your suggestions.

With regards,

embeddedt, 2019-12-16:
> Force Write-Through before enabling DCache

Just a warning: I believe the STM32F746NG is affected by the general Cortex-M7 errata regarding write-through cache (spurious data loaded after a sequence of loads and stores).

bouattay, 2020-01-14:
Hi Desowin,

For USB use cases we recommand even to disable CPU Dcache when USB HS DMA is enabled

Regards,
Aymen

ALABSTM, 2020-03-04:
ST Internal Reference: 82056

ALABSTM, 2020-12-16:
Hi @desowin,

I hope you are fine. Regarding the issue you pointed out, our development teams recommend to use the **DTCM** as this memory is **not cacheable**. Indeed, in case the USB DMA is used, it is mandatory to keep all buffers used by the USB in a non-cacheable region. I hope this helps.

With regards,

TianFuZhang1010, 2023-01-07:
Hi 

I'm using OTG HS on the STM32F746 mcu, in device mode, as a CDC VCP. It works when "Internal IP DMA" is disabled. If I enable DMA, the USB device enumerates but CDC_Transmit_HS() no longer works. 

I saw the article on USB DMA with the STM32H7 and STM32F7, and I tried similar things:

-Disabled the D-Cache globally.
-Edited my linker script to not use DTCM, by skipping past it like this:
"RAM  (xrw)  : ORIGIN = 0x20010000,  LENGTH = 176K"
"RAM_Array_Addr  (xrw) : ORIGIN = 0x2003C000, LENGTH = 64K"  // This is the RAM region for transfer buffer
-I have confirmed that the buffer I am trying to transmit is word aligned (memory address is a multiple of 4.)
-I am trying to transmit 32768 bytes, which is also a multiple of 4.

I would like to use OTG HS with DMA because I need to transfer a lot of data. With DMA disabled, and D-Cache enabled, I am only able to transfer about 20MBps (160Mbps). That is achieved with a loop in main() that non-stop calls CDC_Transmit_HS() to transmit packets that are 32768 bytes long.

I am looking forward to any help.

Thank you,
Tian Zhang.

sistemicorp, 2024-05-09:
I also have this problem, and have gone through many of the steps above.  
I am using STM32H750.  DCcach disabled.  The buffer I have is in RAM_D1.  I am using CDC.
With USB DMA disabled, things work.  With DMA WireShark shows me,
`URB_FUNCTION_SYNC_RESET_PIPE_AND_CLEAR_STALL`
I do get 2-5 trasnfers through, and then it stalls.
I also need DMA because without DMA, the USB HAL driver write function takes too many CPU cycles and my ADC pipelining overruns.  The issue is impacted by how busy Windows host is... when not busy, non-DMA works.  But when my PC side App is very busy, the serial read is slower, so data gets backed up on various levels of buffering.  When PC consumes data gain, STM32H7 tries to send a lot of data back, and those non-DMA transfers take too many cycles.

andy817501, 2024-09-23:
> 我也有这个问题，并且已经完成了上面的许多步骤。 我使用的是 STM32H750。DCcach 已禁用。我的缓冲区在 RAM_D1 中。我正在使用 CDC。 禁用 USB DMA 后，一切正常。WireShark 显示，使用 DMA 后， `URB_FUNCTION_SYNC_RESET_PIPE_AND_CLEAR_STALL` 我确实完成了 2-5 次传输，然后就停滞了。 我还需要 DMA，因为如果没有 DMA，USB HAL 驱动程序写入功能会占用太多 CPU 周期，并且我的 ADC 流水线会超限。该问题受到 Windows 主机繁忙程度的影响……当不忙时，非 DMA 可以工作。但是当我的 PC 端应用程序非常繁忙时，串行读取速度会变慢，因此数据会在各种缓冲级别上备份。当 PC 消耗数据增益时，STM32H7 会尝试发回大量数据，而这些非 DMA 传输需要太多周期。
> 
> 更新：这有帮助， [https://community.st.com/t5/stm32-mcus-embedded-software/problem-with-stm32h747-hs-usb-dma/mp/165464#M9835 ](https://community.st.com/t5/stm32-mcus-embedded-software/problem-with-stm32h747-hs-usb-dma/m-p/165464#M9835) https://community.st.com/t5/stm32-mcus/how-to-correctly-setup-your-application-to-use-usb-dma/ta-p/49522 请注意这一点， `D1 and D2 domains are connected through bus bridges, both can also access data in D3 domain. However, there is no connection from D3 domain to D1 or D2 domain. The DMA1 and DMA2 controllers are located in D2 domain and can access almost all memories with exception of ITCM and DTCM RAM (located at 0x20000000). This DMA is used in most cases. BDMA controller is located in D3 domain and can access only SRAM4 and backup SRAM in D3 domain. MDMA controller is located in D1 domain and can access all memories, including ITCM/DTCM. This controller is mainly used for handling D1 peripherals and memory transfers between domains. In the previous figure, we can see that USBHS1 and USBHS2 are located in D2 domain and have no interconnection with DTCM RAM which is the default memory used in approximately all USB projects. Therefore when enabling the internal DMA USB, projects do not work, as DMA will not be able to access the data buffers placed in DTCM and it results in DMA transfer error.` 并且， `CDC known limitations When using this driver with the OTG HS core, enabling DMA mode (define USB_OTG_HS_INTERNAL_DMA_ENABLED in usb_conf.h file) results in data being sent only by multiple of 4 bytes. This is due to the fact that USB DMA does not allow sending data from non word-aligned addresses. For this specific application, it is advised not to enable this option unless required. From: UM1734 User manual STM32Cube USB device library DocID025934 Rev 1`
> 
> 我还没有达到我想要的最终结果，但我可以与目标进行交换。

I also encountered the same problem, have you solved it?