STMicroelectronics/STM32CubeF7 #9: Order of disabling in `HAL_TIM_IC_Stop_DMA` function

ali-rostami, 2019-12-31:
Hi 
Take a look at [this topic I opened](https://community.st.com/s/question/0D50X0000BudaXESQY/order-of-disabling-in-haltimicstopdma-function?t=1577769025879).

It seems that it's better to change the order of disabling in `HAL_TIM_IC_Stop_DMA` function.
what do you think?

ALABSTM, 2020-01-02:
Hi Ali,

Thank you for reporting this issue. The description on the Forum gives a quite good overview of the case you are describing. However, some details needed by our development teams are still lacking (e.g. firmware revision, IDE, compiler version). Would you mind providing them here?

You could find the exhaustive list in the template that displays when you open a new issue. Thank you in advance for your collaboration.

With regards,

ali-rostami, 2020-01-02:
Hi dear ALABSTM
thanks for your reply. OK, here it is:

**Describe the set-up**
 * The board:  my custom board
 * IDE or at least the compiler and its version: Keil v5.28a, ARM compiler

**Describe the bug**
As described in the previous comment.

**How To Reproduce**
1. Indicate the global behavior of your application project
When I use channel1 of timer1 as the input capture, and also define it as a master of timer3 so then I can get the number of captures by the counter of timer3. there is a difference between the number of captured data and then the counter of timer3.
2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)
The body of `HAL_TIM_IC_Stop_DMA` function.
3. The use case that generates the problem
In the body of `HAL_TIM_IC_Stop_DMA`, as at first, it tries to disable the DMA and then disable the input capture channel, the number of captured data by DMA is fewer than the counter of timer3.

ALABSTM, 2020-01-02:
Hi Ali,

Thank you for all these details. Your issue will be forwarded to our development teams to be analyzed. We will be back to you as soon as we get an answer. Thank you in advance for your patience.

With regards,

FVISTM, 2020-01-15:
Hi Ali,

Thanks for pointing out this issue.
I confirm that TIM capture should be disabled prior stopping the DMA transfers. 
This issue will be fixed in further release of the HAL TIM driver.

FVISTM

ALABSTM, 2020-01-15:
ST Internal Reference: 79347

ali-rostami, 2020-01-15:
> Thanks for pointing out this issue.
> I confirm that TIM capture should be disabled prior stopping the DMA transfers.
> This issue will be fixed in further release of the HAL TIM driver.
> 
> FVISTM

Hi, So great, your welcome. :+1:

ALABSTM, 2021-03-08:
Hi @ali-rostami,

The TIM  HAL driver has been fixed based on what you reported and the new version is now available in the frame of the STM32CubeF7 **v1.16.1** published a couple of days ago. Thank you again for your contribution.

With regards,

closed 2021-03-08 by ALABSTM

ali-rostami, 2021-03-08:
Hi dear @ALABSTM 
Glad to hear that. thanks🙏🎉
 I almost forgot it. -As you know it took more than a year to be fixed.😉-
Anyway, I hope this process will get faster by letting the STM32 users send PR. which will get us closer too.🙂

Thanks again 
Best regard

ALABSTM, 2021-03-23:
Dear @ali-rostami,

Good to read your comment.

>Anyway, I hope this process will get faster by letting the STM32 users send PR. which will get us closer too.🙂

Regarding what you mentioned about pull-requests, we opened the door to such contributions about a year ago. We are looking forward to receiving yours soon.

With regards,

ali-rostami, 2021-04-03:
> Regarding what you mentioned about pull-requests, we opened the door to such contributions about a year ago. We are looking forward to receiving yours soon.

Glad to hear that. 
thanks 🎉