STMicroelectronics/STM32CubeF7 #13: stm32f7xx_hal_rcc.c - Assert Failed

svcguy, 2020-01-30:
**Describe the set-up**
Custom Board with STM32F767IG
STM32CubeF7 FW 1.15.0
STM32CubeIDE 1.2.1

**Describe the bug**
Assert Failed - FILE: stm32f7xx_hal_rcc.c LINE: 602
`RCC_OscInitStruct.PLL.PLLR` is not initialized to a value and must be `>=2 && <=7`

**How To Reproduce**
[See attached *.ioc](https://github.com/STMicroelectronics/STM32CubeF7/files/4134364/genericDisplay.ioc.txt)

**Additional context**
Code below fixes the problem:
```
/* CUBEMX BUG FIX
  * PLLR not initialized to a valid value
  * PLLR >=2 && <= 7
  * Initialize to 2 (reset value)
  */
RCC_OscInitStruct.PLL.PLLR = 2;
```
The dropdown for the R divider on PLL is greyed out in the clock configuration:
[image]

Posted on [community.st.com](https://community.st.com/s/question/0D50X0000C1ziS5SQI/bug-cubemx-stm32f767ig-rcc-assert-failed) as well.

svcguy, 2020-01-30:
Upon further review, I’m guessing this is more of a CubeMX bug than a HAL bug.  Feel free to close if that’s the case.

ALABSTM, 2020-02-03:
Hi Andy,

Thank you for this report. May I ask you the Cube MX revision you are using?

With regards,

svcguy, 2020-02-03:
5.5.0

ALABSTM, 2020-02-05:
Hi Andy,

Thank you for this report, once again. However, aspects related to Cube MX tool are not handled at GitHub level. I do apologize for the confusion that may happened after my previous comment, but we are not going to handle this issue.

It may rather be treated at **ST Community** level, precisely within the [Cube MX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated section. There, you may find posts that already raised the issue you are pointing out here. Otherwise, you can create a new post to submit it.

Please allow me to close this issue. Thank you for your comprehension and my apologies once more.

With regards,

closed 2020-02-05 by ALABSTM