STMicroelectronics/STM32CubeF7 #14: Build and run an example in atollic TrueSTUDIO (or any other IDE)

franklevasseur, 2020-02-04:
Hi,

I need some help here,

I'm currently trying to simply import one of the examples in attolic TrueSTUDIO, build it and run it on my STM32F767Zi Nucleo development kit. To say the least, I'm having a hard time. There's no specific information on this topic in all this repo's readme files. 

I know this issue is not exactly related to this specific repo, but I feel like code examples should come with some sort of tutorial to build and run them on a dev kit.

The only instructions I could find one this repo are:

"In order to make the program work, you must do the following :
 - Open your preferred toolchain 
 - Rebuild all files and load your image into target memory
 - Run the example"

When I click "Import Project from File System of Archive", and I select an example, ([this one](https://github.com/STMicroelectronics/STM32CubeF7/tree/master/Projects/STM32F767ZI-Nucleo/Examples/ADC/ADC_RegularConversion_Polling)) that I have previously downloaded, the "build project" option is greyed out:

[image]

I also would like a solution that does not imply that I create a project that builds and run and then replace all source files with the ones from this [example](https://github.com/STMicroelectronics/STM32CubeF7/tree/master/Projects/STM32F767ZI-Nucleo/Examples/ADC/ADC_RegularConversion_Polling)...

Any help would be appreciated.

Sorry in advance for my bad english.

François

ALABSTM, 2020-02-05:
Hi François,

Please allow me to introduce @tarek-bochkati who will help you deal with this "hard time" you are facing.

Please allow me also to kindly remind that issues are particularly intended to report problems related to the software **published within these GitHub repositories**. Any other topic (tools, boards, chips, etc.) shall be addressed to the ST Community.

Issues related to "True Studio", for instance, can be posted [here](https://community.st.com/s/topic/0TO0X000000BWSuWAO/truestudio). This other [page](https://community.st.com/s/topiccatalog) lists the different topics available. Thank you for your comprehension.

With regards,

tarek-bochkati, 2020-02-05:
Hi François,

TrueSTUDIO for STM32 does not have an intuitive way to import SW4STM32 projects,
but in emcu.eu there is a [tutorial ](http://www.emcu.eu/how-to-import-in-atollic-truestudio-for-stm32-9-0-0-a-project-developed-for-ac6-system-workbench/) explaining "**How to import in Atollic TrueSTUDIO for STM32 a project developed for AC6 (System WorkBench)**"

BTW, **TrueSTUDIO** is replaced by **STM32CubeIDE** which offers an easy import feature.
_We strongly recommend to start using STM32CubeIDE as TrueSTUDIO has reached his end of life._

Kind Regards,
Tarek

franklevasseur, 2020-02-05:
Hi Tarek,

Thank you for your time and tutorial recommendation!

I will start using STM32CubeIDE. and, for more questions, I will post on the community forum.

I'm so new to embedded development that I had not enough vocabulary to correctly formulate my question. **SW4STM32** and **AC6 (System WorkBench)** are keywords I didn't even know...

Many thanks! 

françois :)

tarek-bochkati, 2020-02-05:
No worries :)
SW4STM32 is an abbreviation for "System Workbench for STM32", it is the IDE as AC6 (System WorkBench)

ALABSTM, 2020-02-07:
Hi guys,

@franklevasseur, Glad that you found answers for your questions. @tarek-bochkati, thank you very much for your support. @franklevasseur, could this issue be closed?

With regards,

franklevasseur, 2020-02-09:
> Hi guys,
> 
> @franklevasseur, Glad that you found answers for your questions. @tarek-bochkati, thank you very much for your support. @franklevasseur, could this issue be closed?
> 
> With regards,

Yes! Sorry for the delay...

I was able to build and run an example without problem using the tutorial you recommended.

This issue can be closed.

François :)

closed 2020-02-10 by ALABSTM