STMicroelectronics/STM32CubeF7 #15: Excess garbage.

AVI-crak, 2020-02-27:
Enumeration of position and mask for individual bits of the integer 32b bit parameter in CAN devices.
For every single bit there are "_Pos" and "_Msk" - but that doesn't make sense. For example, for CAN_F2R2. Here, and in many other places.
I understand that it is important for you to maintain compatibility with older versions, but what to do if no one uses this code (even crazy geeks).

Cleaning your trash will allow you to use free space for more important information. For example, put one file everything related to the periphery, except for the executable code. Users are already accustomed to the large file size, and are able to use the search (tooltip).

ALABSTM, 2020-03-30:
Hi AVI-crak,

I hope you are doing well. Regarding the point you raised, you may have noticed that each one of the bits you are referring to (and many others too) is listed in the reference manual. As so, it has to be reported into the CMSIS Device header file.

Now, regarding the use of the "_Pos" and "_Msk" formalism, as you said, this is for the sake of compatibility and standardization requirements. Such requirements are major ones, even though complying to them may yield numerous lines of code.

From your side, why not creating your own customized header file? From ours, we have to ensure to all our users a minimal form of standardization and, thus, cannot apply different rules to different cases, especially within the CMSIS Device header file, the coding rules of which are suggested by Arm itself.

I hope this answers (even a little bit) your question. Thank you for your comprehension and thank you for having tackled the point too. Take care and stay safe.

With regards,

AVI-crak, 2020-03-30:
Have a nice one you too.

Do some of the work yourself? Yes, I’m just doing this. Because ready-made solutions are too bulky and massive. I collect code examples for my tasks, in a simplified form, in many versions, for the purpose of reuse in my projects. All code is based on direct access to registers, and these are not ready-made functions, but pieces of code for copying into a project. And this approach saves a lot of time.

However, it does not make sense to demonstrate the results of the work. This is part of the path of another road, my personal rut. The chance to use my path by other players is almost zero. It's all about psychology, players are much more comfortable using that tool - which is the standard.

About the rules. My suggestion is to change the rules.
All that is a whole bit field is to use as a whole field, without describing the individual bits of this field. Register CAN_F2R2 in this regard as the most striking version of the result of the rule change. Instead of 95 lines, only three lines remain.

Changing the rules will require changing the standards, to simpler and more understandable ones. In this case, you can temporarily leave two branches of development, and monitor the players. As soon as the old code becomes out of date, freeze the old standard.
I think that in reality this will happen quickly enough, because almost any project starts from scratch.

ALABSTM, 2020-04-02:
Hi @AVI-crak,

As your request cannot be fulfilled, please allow me to close this issue now. Thank you again for your comprehension.

With regards,

closed 2020-04-02 by ALABSTM