STMicroelectronics/STM32CubeF7 #17: Сode optimization problem in stm32f7xx_hal_spi.c

alekseyzubk0v, 2020-04-21:
IAR 7.5 deletes this code on any optimization settings.

https://github.com/STMicroelectronics/STM32CubeF7/blob/08376dce1b404687e4a86b73077f396bccfc9cb5/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_spi.c#L3847

ASELSTM, 2020-04-24:
Hi @alekseyzubk0v,

Thank you for your contribution.
Have you tried to check with more recent IAR versions whether it is also optimized or not ?

With regards,

alekseyzubk0v, 2020-04-25:
Hi  @ASELSTM,
No, I did not try to check with a more recent version of IAR.

ASELSTM, 2020-05-12:
Hi @alekseyzubk0v ,

The issue you pointed out have been transmitted to our development teams, and they confirmed it.
A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-05-12:
ST Internal Reference: 86220

ASELSTM, 2021-03-04:
Hi @alekseyzubk0v,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.16.1 of the STM32CubeF7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-04 by ASELSTM