STMicroelectronics/STM32CubeF7 #18: QSPI_ReadWrite failed to verify written date

banxian, 2020-06-07:
Hardware: STM32F723E-Discovery
Software: STM32Cube_FW_F7_V1.15.0/STM32Cube_FW_F7_V1.15.0
IDE: EWARM v8, MDK-ARM v5

Step to reproduce:
open project under MDK-ARM folder, build and debug.
place a breakpoint on  "if(Buffercmp(qspi_aRxBuffer, qspi_aTxBuffer, BUFFER_SIZE) > 0)"
when breakpoint hits, add watch on qspi_aRxBuffer.
the content in qspi_aRxBuffer is 00,10,11,00, 00,14,11,00, 00,18,19,00, 00,1C,19,00
seems it lost certain bits from original buffer 0F,10,11,12, 13,14,15,16.

I tried modify WRITE_READ_ADDR to 0x200 to avoid page align algorithm, it takes no help.
turn off compiler's optimization takes no help.

EWARM have same corrupted content in qspi_aRxBuffer.

RKOUSTM, 2020-06-08:
Hi @banxian, 

Thank you for your contribution. Could you please confirm you are using the latest firmware version (STM32Cube F7 v1.16.0) and that the issue is systematically reproducible on your side?

With regards,

RKOUSTM, 2020-06-15:
Hi @banxian,

The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

RKOUSTM, 2020-06-16:
ST Internal Reference: 88376

RKOUSTM, 2020-12-22:
Hi @banxian,

Thank you for your contribution.  Indeed, the behavior of [BSP exemple](https://github.com/STMicroelectronics/STM32CubeF7/tree/master/Projects/STM32F723E-Discovery/Examples/BSP) is aligned with the readme.txt description for all optimization levels (None, Medium, Low and High). It seems that the content of the Tx/Rx buffer is wrong because you are debugging with a high optimization level, where it is not recommended to debug with this level and proper operation is not guaranteed. In general, using a high optimization level may affect the code in a way that will not allow you to view a value of a variable as expected.

Please let us know your feedback so that we can help you. Thank you again for your contribution.

With regards,

RKOUSTM, 2021-02-09:
Please allow me to close this thread as no activity has been registered since two months. You can reopen it at any time if you have details to provide us to help you solve this issue.
Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-02-09 by RKOUSTM