STMicroelectronics/STM32CubeF7 #19: Fatfs - SD card - gives Disk-Not-Ready error during subsequent f_mount()

alan-wood, 2020-07-11:
**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
  Custom board. STM32F746 with SD Card interface, USB, DCMI .
 * IDE
  AC6 SystemWorkbench ( eclipse )
  gcc compiler 
  CubeMX 5.6.1
  HAL 1.16.0
  FatFS

  

**Describe the bug**
 When trying to initialise the FAT filesystem ( f_mount() ) I would sometimes get an error of DISK NOT READY.
 This rarely ( if ever ) happened on the first initialisation after power up and did not happen if all we did was loop around a f_mount, and unmount.
 But if we ran the rest of our system which involved mounting the disk, reading/writing files and then unmounting the DISK NOT READY would occur.

**How To Reproduce**
1. Indicate the global behavior of your application project.
 In my project running the equivalent of an 'ls' command where the code mounts the drive, reads the directory and then unmounts the drive.
 This would often fail within 5 attempts.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
  stm32f7xx_hal_sd.c : SD_FindSCR()

3. The use case that generates the problem.
 I am a little unsure of the complete case here but the I have seen SDMMC_FLAG_DBCKEND set on entry to this routine which upsets the code. 

4. How we can reproduce the problem.
 

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.
 The analysis has been:
 Tracing the case of the Not Ready Error back to SD_FindSCR() returning 0,0 for the SCR register. 

My fix for this is to add the line:
  __HAL_SD_CLEAR_FLAG(hsd, SDMMC_STATIC_DATA_FLAGS);
before:
  /* Send ACMD51 SD_APP_SEND_SCR with argument as 0 */
  errorstate = SDMMC_CmdSendSCR(hsd->Instance);

RKOUSTM, 2020-07-13:
Hi @alan-wood,

Thank you for this report. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get feedback.

Thank you for your patience and thank you again for your contribution.

With regards,

RKOUSTM, 2020-07-28:
Hi @alan-wood,

The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards

RKOUSTM, 2020-07-28:
ST Internal Reference: 90556

RKOUSTM, 2021-03-03:
Hi @alan-wood,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.16.1 of the STM32CubeF7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-03 by RKOUSTM