STMicroelectronics/STM32CubeF7 #20: SDMMC_DATATIMEOUT is used in multiple ways in the code and thus can't be adjusted.

alan-wood, 2020-07-11:
**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
  Custom board. STM32F746 with SD Card interface, USB, DCMI .
 * IDE
  AC6 SystemWorkbench ( eclipse )
  gcc compiler 
  CubeMX 5.6.1
  HAL 1.16.0
  FatFS

**Describe the bug**
Clean up the use of SDMMC_DATATIMEOUT.

**How To Reproduce**
 Code Review:
 In stm32f7xx_hal_sd.c the value SDMMC_DATATIMEOUT is used in two ways.
 The first is as a value for the SDMMC.DTIMER register which has units of the SD Clock ( 48Mhz for me ).
 The other uses has units of the HAL_GetTick() ( 1ms ).

 While trying to adjust the later value to give up after a few seconds with SDMMC_DATATIMEOUT=3000.
 The SD Card was not completing the transfer in the 62us, giving a data timeout from the SDMMC.

Suggestion:
 Use separate variables for the two uses.

ASELSTM, 2020-07-13:
Hi @alan-wood,

Thank you for reporting this report. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get feedback.

Thank you for your patience and thank you again for your contribution.

With regards,

ASELSTM, 2020-07-14:
Hi @alan-wood,

Our development teams confirmed your request. Actually, the point you raised has also been discussed internally since a while, before we receive your request. A fix will be made available in the frame of a future release.

We cannot share a date for the moment. Hopefully, this should be done soon. So stay tuned and thank you once more for your contribution and for your patience.

With regards,

ASELSTM, 2020-07-14:
ST Internal Reference: 43235

ASELSTM, 2021-03-04:
Hi @alan-wood,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.16.1 of the STM32CubeF7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-04 by ASELSTM

gcmitec, 2021-03-12:
Hi ASELSTM,
 are you sure?
https://github.com/STMicroelectronics/STM32CubeF7/blob/c7c5ec99c7482ea8bcdbf0a869c930af4547088f/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_sd.c#L603
https://github.com/STMicroelectronics/STM32CubeF7/blob/c7c5ec99c7482ea8bcdbf0a869c930af4547088f/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_sd.c#L2868