STMicroelectronics/STM32CubeF7 #21: SDMMC FIFO not being emptied in SD_FindSCR

alan-wood, 2020-07-11:
**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
  Custom board. STM32F746 with SD Card interface, USB, DCMI .
 * IDE
  AC6 SystemWorkbench ( eclipse )
  gcc compiler 
  CubeMX 5.6.1
  HAL 1.16.0
  FatFS

**Describe the bug**
 While trying to fix the issue described in Issue #19 and understand the SDMMC module in the STM32F746 I believe that there is an issue with the reading of the SDMMC_FLAG_DBCKEND in relation to the SDMMC_FLAG_RXDAVL flag in theSD_FindSCR function.
 The STM32F74 reference manual is not clear on the relationship between the status flags for the DPSM and the FIFO and I have understood from the reference manual that the DBCKEND flag can be set when the FIFO still contains data that has not been read.
 If that is the case then the existing code will not work correctly when the processor is slowed down ( through low clock speed or interrupt load ).

 The HAL_SD_ReadBlocks() function uses the SDMMC_FLAG_DATAEND flag rather than SDMMC_FLAG_DBCKEND and then continues to empty the FIFO.
 SD_SendSDStatus() seems to have similar code to SD_FindSCR but it continues to empty the FIFO after the DBCKEND flag.

**How To Reproduce**

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
  stm32f7xx_hal_sd.c : SD_FindSCR()

3. The use case that generates the problem.
  I have not been able to confirm my suspicions here.

4. How we can reproduce the problem.
 

**Additional context**

My fix for this is to test for SDMMC_FLAG_DBCKEND only after SDMMC_FLAG_RXDAVL has been fully serviced:

  while(!__HAL_SD_GET_FLAG(hsd, SDMMC_FLAG_RXOVERR | SDMMC_FLAG_DCRCFAIL | SDMMC_FLAG_DTIMEOUT ))
  {
    while(__HAL_SD_GET_FLAG(hsd, SDMMC_FLAG_RXDAVL))
    {
      tempscr[index++] = SDMMC_ReadFIFO(hsd->Instance);
    }

    if (__HAL_SD_GET_FLAG(hsd, SDMMC_FLAG_DBCKEND) && !__HAL_SD_GET_FLAG(hsd, SDMMC_FLAG_RXDAVL))
    {  // end of data transfer from Card and from FIFO
    	break;
    }

    if((HAL_GetTick() - tickstart) >=  SDMMC_DATATIMEOUT)
    {
      return HAL_SD_ERROR_TIMEOUT;
    }
  }

I think that the non-interrupt data read function should be factored out of the three routines ( SD_SendSDStatus, SD_FindSCR, HAL_SD_ReadBlocks ).

ASELSTM, 2020-07-13:
Hi @alan-wood,

You request will be forwarded to our development teams. I will let you updated once they provide me with their answer.

With regards,

ASELSTM, 2020-09-24:
Hi @alan-wood,

As promised we have reported your report to our development teams and they confirmed the issue. A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-09-24:
ST Internal Reference: 93301

ASELSTM, 2021-03-04:
Hi @alan-wood,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.16.1 of the STM32CubeF7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-04 by ASELSTM