STMicroelectronics/STM32CubeF7 #22: Compilation error in STM32CubeF7\Projects\STM32F7508-DISCO\Demonstrations\STemWin

navinbkhatri, 2020-07-29:
these demonstration code should compile and run without making any changes in the code, but unfortunately there are some errors that i found in these demo code, i hope it will be resolved.
error_1: "no source": Error:  #5: cannot open source input file "../../../../../Middlewares/ST/STemWin/OS/GUI_X_OS.c": No such file or directory
error_2: ../Gui/Target/GUIConf.c(54): error:  #5: cannot open source input file "GUI.h": No such file or directory.
I am new to STM32F7508 programming please suggest where can i find these GUI files.

RKOUSTM, 2020-08-10:
Hi @navinbkhatri,

Thank you for your report. A fix will be implemented and made available in the frame of a future release.

We cannot share a date for the moment, but hopefully it should be soon. Thank you for your patience and thank you once more for your contribution.

With regards,

RKOUSTM, 2020-08-11:
Hi @navinbkhatri,

Welcome to our GitHub organization. In fact, some Middlewares are not directly available on GitHub as they are subject to a restrictive license (ST Ultimate Liberty). To solve your problem you just need to download
the STemWin module in this directory ``(STM32CubeF7/Middlewares/ST/STemWin/)``. A [link](https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Middlewares/ST/STemWin/readme.txt) will redirect you to the MW web page on ST servers to download the STemWin module. 

With regards,

closed 2020-08-17 by RKOUSTM