STMicroelectronics/STM32CubeF7 #23: uwTickPrio not set to TickPriority in timer based HAL_InitTick

x893, 2020-08-11:
Hi,
stm32f7xx_hal.c
__weak HAL_StatusTypeDef HAL_InitTick(uint32_t TickPriority)
contain next code
  /* Configure the SysTick IRQ priority */
  if (TickPriority < (1UL << __NVIC_PRIO_BITS))
  {
    HAL_NVIC_SetPriority(SysTick_IRQn, TickPriority, 0U);
    uwTickPrio = TickPriority;
  }
  else
  {
    return HAL_ERROR;
  }

but stm32f7xx_hal_timebase_tim.c HAL_StatusTypeDef HAL_InitTick (uint32_t TickPriority) not contain and HAL_RCC_ClockConfig(...) fail with full assert.

RKOUSTM, 2020-08-14:
Hi @x893,

The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

RKOUSTM, 2020-08-14:
ST Internal Reference: 69397

RKOUSTM, 2021-03-03:
Hi @x893,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.16.1 of the STM32CubeF7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-03 by RKOUSTM