STMicroelectronics/STM32CubeF7 #25: Remove deprecated `register` keyword

ladislas, 2020-09-13:
**Describe the set-up**

 * DISCO F769
 * gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)

**Describe the set-up**

* DISCO F769
* gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)

**Describe the bug**

Compiling code with `-std=gnu++17` and later, gcc warns about the following:

```console
warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
```

**How To Reproduce**

Compile hal with -std=gnu++2a or -std=gnu++17

**Additional context**

The issue has been brought up in multiple places and the conclusion is that the removal of the `register` keyword (deprecated since 2009) doesn't have any impact.

Discussions and details here:

- https://github.com/STMicroelectronics/STM32CubeF4/issues/27
- https://github.com/ARM-software/CMSIS_5/issues/345
- https://github.com/ARM-software/CMSIS_5/pull/365
- https://github.com/ARM-software/CMSIS_5/pull/347#issuecomment-388333964
- https://github.com/ARMmbed/mbed-os/issues/12232

ASELSTM, 2020-09-16:
Hi @ladislas,

First of all we would like to thank you for your contribution.

Actually, the point you raised is already known. It has been internally fixed. The fix will be published in the frame of a future publication. We cannot share a date for the moment. Stay tuned and thank you once more for your contribution and for your patience.

With regards,

ASELSTM, 2020-09-16:
ST Internal Reference: 84397

ladislas, 2020-09-16:
Good to hear! Can't wait to see the update coming out.

In the mean time, using `-Wno-register` disables the warning.

ALABSTM, 2020-10-27:
Duplicate of [STM32CubeF4#27](https://github.com/STMicroelectronics/STM32CubeF4/issues/27).

ASELSTM, 2021-03-04:
Hi @ladislas,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.16.1 of the STM32CubeF7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-04 by ASELSTM