STMicroelectronics/STM32CubeF7 #26: STM32F769I-Discovery - BSP example does not work - DFSDM/SAI problem

AleksMx, 2020-09-16:
**Caution**
I downloaded the latest version of STM32CubeF7. Then I opened _STM32CubeF7-master\Projects\STM32F769I-Discovery\Examples\BSP\SMDK-ARM_ and compiled project with uVision Keil 5.31 MDK-ARM. The project was compiled without any errors. Then I used STM32 Utility to write sound to the memory 0x08080000.

But **audio_play** and **audio_loopback** examples reproduce strange iron sound.

**Describe the set-up**
STM32F759I-DISCO with LCD screen. 

**Describe the bug**
Can't hear the right sound.

**How To Reproduce**
1. Download master version of STM32CubeF7
2. Build with uVision Keil
3. Press User button to select audio_loopback example (LCD works well)

**Additional context**
I'm having big problems with this board. I was able to build project in Atollic Studio which use audio_loopback example (from master branch) and plays sound successfully. But it play sound successfully only if I init _USART6_ by calling MX_USART6_UART_Init method after call BSP_AUDIO_IN_Init and before call BSP_AUDIO_OUT_Init and only 16 kHz... It sounds strange, but it is. But when I started USART6 to receive data by calling HAL_UART_Receive_DMA method - DFSDM module stopped sending data.

After that I decided to compile the example from STM32CubeF7 and I faced with this problem.

Also I have tested examples from STM32CubeF7-1.15.0 - the result is the same (problem with audio_loopback example).
I have tested with another power cable, power supply - the result is the same.

I do not know what to do. I have a deadline soon. And I can not get to work an elementary example.

RKOUSTM, 2020-09-22:
Hi @AleksMx,

Thank you for pointing out this issue. It has been transferred to our development teams for deeper analysis. We will be back to you as soon as we get an answer. Thank you for you patience.

With regards

RKOUSTM, 2020-09-25:
Hi @AleksMx,

Thank you for your contribution.  To remove the metallic noise and to hear the audio file,  the modification below in the target options (``IROM1`` and ``IRAM1``) is proposed: 

[image]

In order to remove the white noise and to enhance the quality of the sound, you can modified the BSP audio frequency from ``44K`` to ``32K ( or less)`` in the ``AudioLoopback_demo ()`` as shown below: 
```patch
@@ -97,2 +97,2 @@ /* Initialize Audio Recorder with 4 channels to be used */
- if (BSP_AUDIO_IN_Init(BSP_AUDIO_FREQUENCY_44K, DEFAULT_AUDIO_IN_BIT_RESOLUTION, 2*DEFAULT_AUDIO_IN_CHANNEL_NBR) == AUDIO_OK)
+ if (BSP_AUDIO_IN_Init(BSP_AUDIO_FREQUENCY_11K, DEFAULT_AUDIO_IN_BIT_RESOLUTION, 2*DEFAULT_AUDIO_IN_CHANNEL_NBR) == AUDIO_OK)
```
```patch
@@ -128,4 +128,4 @@ /* Initialize the audio device*/ 
- if (BSP_AUDIO_OUT_Init(OUTPUT_DEVICE_HEADPHONE, 
- 70, 
-  BSP_AUDIO_FREQUENCY_44K) != AUDIO_OK)
+ if (BSP_AUDIO_OUT_Init(OUTPUT_DEVICE_HEADPHONE, 
+ 70, 
+ BSP_AUDIO_FREQUENCY_11K) != AUDIO_OK)
```
Thank you again for your contribution.

With regards,

RKOUSTM, 2020-09-28:
ST Internal Reference: 93452

RKOUSTM, 2020-09-28:
ST Internal Reference: 93458

ALABSTM, 2023-04-04:
Hi,

Fixed in version **1.16.2** of the STM32CubeF7 firmware.

With regards,

closed 2023-04-04 by ALABSTM