STMicroelectronics/STM32CubeF7 #27: __HAL_RCC_USB_OTG_FS_CLK_ENABLE: declaration of 'tmpreg' shadows a previous local

daniel-starke, 2020-09-30:
**Describe the bug**
The macro `__HAL_RCC_USB_OTG_FS_CLK_ENABLE`
https://github.com/STMicroelectronics/STM32CubeF7/blob/08376dce1b404687e4a86b73077f396bccfc9cb5/Drivers/STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal_rcc_ex.h#L845-L852
declares the local variable `tmpreg` and calls the macro `__HAL_RCC_SYSCFG_CLK_ENABLE` which is defined as
https://github.com/STMicroelectronics/STM32CubeF7/blob/08376dce1b404687e4a86b73077f396bccfc9cb5/Drivers/STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal_rcc.h#L468-L474
and also declares a local variable with the name `tmpreg`. This shadows the first declaration of the variable. I suggest the rename `tmpreg` in  `__HAL_RCC_USB_OTG_FS_CLK_ENABLE` to something else.

**How To Reproduce**
Use `__HAL_RCC_SYSCFG_CLK_ENABLE()` somewhere in the source with the compiler flag `-Wshadow` set.

ALABSTM, 2020-10-14:
Hi @daniel-starke,

Thank you for this report. Unfortunately, we could not reproduce the issue you described. Would you mind indicating the compiler you are using and its version? Any other detail on how to reproduce this warning is also welcome.

Thank you,

ALABSTM, 2020-10-14:
Hi @daniel-starke,

According to our analysis, each one of the `tmpreg` has a distinct scope from the other one thanks to the `do { ... } while(0)` blocs surrounding each variable. We performed a couple of tests using the `printf()` function to print the value stored in the `tmpreg` variables and noticed no impact of the variable shadowing.

Moreover, the `tmpreg` variables are not meant to be used. This is the purpose from the `UNUSED(tmpreg);` lines in both macros. The read operation is only intended to create a delay as indicated in the comment.

Please do not hesitate to share your point of view. Any other detail on how to reproduce this warning is also welcome.

With regards,

daniel-starke, 2020-10-14:
Yes, there is no functional impact. This is just an aesthetic issue due to the warning from the compiler which makes it harder to see real issues.

carlosgalvezp, 2020-10-14:
@daniel-starke I fully agree this should be solved. However it's hard to expect 3rd-party libraries to fix stricter warnings.

You can include the headers with -isystem to let the compiler ignore those warnings, so you can still have your strict warnings active.

daniel-starke, 2020-10-14:
True. But even if no fix is provided, reporting it may help others to evaluate their findings.

ALABSTM, 2020-10-15:
Hi,

@carlosgalvezp, thank you for your comment. @daniel-starke, indeed, it is interesting to have this point reported whether for other users' attention or even for us at STMicroelectronics. Thank you again.

With regards,

ALABSTM, 2020-10-27:
Hi,

Please allow me to close this issue. You can re-open it at any time in case any noticeable side effect is experienced due to this variable shadowing.

With regards

closed 2020-10-27 by ALABSTM