STMicroelectronics/STM32CubeF7 #28: Memory corruption

GiulioDallaVecchia, 2020-10-19:
**Issue**
I'm working on my custom board with STM32F7, where I'm using the SBSFU as bootloader. When I use the FPU on the application (on the assembly istruction: VMOV.F32 #s0, #0.25) and the FPU has already been used in the bootloader side a corruption memory happens with this pattern (at address 0x200152C): 0x00 0x00 0x00 0x40 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xFF 0xFF 0xFF 0xFF 0x00 0x00 0x00 0x80.

It seems to be a bug. Is there any raccomandation to use the FPU on the bootloader side before to jump to application side? I have not found anythings about that.

**Describe the set-up**
 * STM32F746VGT6
 * Keil uVision

ALABSTM, 2021-01-19:
Hi @GiulioDallaVecchia,

Thank you for your question. According to our development teams, the user application and the secure boot are two **different** and **independent** binaries. Their recommendation is the following:
* Since secure boot does not require FPU, SBSFU shall be built with FPU **disabled**.
* As the applications requires FPU, it shall be built with this latter **enabled**.
We hope this help. Do not hesitate in case you still have questions about the subject.

We do apologize for the delay it may take sometimes to reply. Thank you for your patience.

With regards,

ALABSTM, 2021-02-22:
Hi @GiulioDallaVecchia,

I hope you are fine. Any feedback after my last comment?

Thanks,

GiulioDallaVecchia, 2021-02-23:
Hi @ALABSTM,

I apologize for the delay. I have tried to disable FPU on the SBSFU project and the problem disappeared. Nevertheless, I think it's quite strange that if I enable FPU on the bootloader side when I reuse it on the main application a memory corruption appears. Have you ever seen something like that?

Best regards,

atsju, 2021-08-16:
Hi @GiulioDallaVecchia 
I'm not from ST, just randomly reading the interesting issues.
I agree with you ST should give more explaination than just "disable FPU in bootloader".
Maybe this could help : https://community.st.com/s/question/0D50X0000BD2Fuv/mpu-configuration-in-combination-with-sbsfu-secure-boot-and-secure-firmware-upgrade-package The explaination is very clear and I really think it could be linked to your issue.

Here is the cortex M7 FPU documentation : https://developer.arm.com/documentation/dui0646/a/cortex-m7-peripherals/floating-point-unit?lang=en

GiulioDallaVecchia, 2021-08-23:
Thank a lot for sharing @atsju!

ALABSTM, 2022-02-14:
Hi,

@atsju, thank you for having shared your findings.

@GiulioDallaVecchia, as this issue solved, please allow me to close it. Thank you.

With regards,

closed 2022-02-14 by ALABSTM