STMicroelectronics/STM32CubeF7 #29: uint8_t BSP_AUDIO_IN_OUT_Init(uint16_t InputDevice, uint16_t OutputDevice, uint32_t AudioFreq, uint32_t BitRes, uint32_t ChnlNbr)

nusret103, 2020-10-21:
Dear,

When i call the function

```
uint8_t BSP_AUDIO_IN_OUT_Init(uint16_t InputDevice, uint16_t OutputDevice, uint32_t AudioFreq, uint32_t BitRes, uint32_t ChnlNbr)
```
with parameters
```
inputDevice == INPUT_DEVICE_INPUT_LINE_1

(BSP_AUDIO_IN_OUT_Init(inputDevice, OUTPUT_DEVICE_HEADPHONE,
                                      FS_, DEFAULT_AUDIO_IN_BIT_RESOLUTION,
                                      DEFAULT_AUDIO_IN_CHANNEL_NBR)
```
I am getting an error can you let me know what is the issue?

Nusret

**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.

**Additional context**
If you have a first analysis or a patch proposal, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

RKOUSTM, 2020-10-27:
Hi @nusret103,

Thank you for your contribution.

Actually, the point you raised has already been raised in another issue #26. It has been internally fixed. The fix will be published in the frame of a future publication.

With regards,

RKOUSTM, 2020-10-27:
ST Internal Reference: 93452

RKOUSTM, 2020-10-27:
ST Internal Reference: 93458

nusret103, 2020-10-27:
@RKOUSTM Thank you for your feedback.

Did you have any time frame when this issue will be ready for testing?

With respect,

RKOUSTM, 2020-10-27:
Hi @nusret103,

Thank you for your contribution. 
This update will be integrated in the future firmware package release (STM32F7 version v1.16.0)  firmware package.
Stay tuned.

Thank you again for your contribution.

With regards,

closed 2022-02-23 by RKOUSTM