STMicroelectronics/STM32CubeF7 #30: USBD_LL_GetRxDataSize/HAL_PCD_EP_GetRxCount return wrong value

davekeck, 2020-10-26:
When both of these conditions are true:

  - A USB OTG HS device has DMA enabled ("Enable internal IP DMA" checkbox in the UI)
  - `USBD_LL_PrepareReceive` is used to initiate a transfer that's larger than the max packet size

Under these circumstances, when the DataOut USB callback is called as a result of `USBD_LL_PrepareReceive`, `USBD_LL_GetRxDataSize`/`HAL_PCD_EP_GetRxCount` return an incorrect value for the number of bytes received.

The bug appears to be due to these lines in `PCD_EP_OutXfrComplete_int`:

https://github.com/STMicroelectronics/STM32CubeF7/blob/08376dce1b404687e4a86b73077f396bccfc9cb5/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal_pcd.c#L1986-L1990

Th above math incorrectly uses the the max packet size (instead of the transfer size), so that when the transfer size is larger than the max packet size, `xfer_count` is assigned to an incorrect value, which breaks `USBD_LL_GetRxDataSize`/`HAL_PCD_EP_GetRxCount`.

This replacement code fixes the issue for me:

```c
const uint32_t xfer_len = hpcd->OUT_ep[epnum].xfer_len;
const uint32_t xfrsiz = (USBx_OUTEP(epnum)->DOEPTSIZ & USB_OTG_DOEPTSIZ_XFRSIZ);
if (xfer_len) {
    const uint32_t xfer_count = xfer_len - xfrsiz;
    hpcd->OUT_ep[epnum].xfer_count = xfer_count;
    hpcd->OUT_ep[epnum].xfer_buff += xfer_count;
} else {
    hpcd->OUT_ep[epnum].xfer_count = 0;
}
```

ALABSTM, 2020-12-15:
Hi @davekeck,

Thank you for this clear and concise report and for the fix proposal. It has been forwarded to our development teams. I will get to you as soon as they provide their feedback.

With regards,

ALABSTM, 2022-01-14:
ST Internal Reference: 120891

closed 2022-01-14 by ALABSTM