STMicroelectronics/STM32CubeF7 #33: Can't configure PLL on STM32F722 with CubeF7 v1.16

elgarbe, 2020-11-11:
I'm using STM32CubeMX 6.0.1 with STM32CubeF7 v 1.16 in order to code my application.
I just config HSI + PLL in order to obtain 216MHZ. I'm not configuring other things, just the Clock
I'm using SW as the IDE. I build the project and start a debug session.
As soon as the main program start (breakpoints on HAL_Init) I evaluate:
__HAL_RCC_GET_SYSCLK_SOURCE() and get 8 (instead of 0)
RCC->CFGR and get 37898 (instead of 0)

With this values HAL_RCC_OscConfig returns an HAL_Error

This problem is not present on STM32CubeF7 v1.12 + CubeMX v4.27

[test_CLK_v116.ioc.zip](https://github.com/STMicroelectronics/STM32CubeF7/files/5521372/test_CLK_v116.ioc.zip)

elgarbe, 2020-11-11:
Can someone confirm this?

elgarbe, 2020-11-12:
this issue is not present on CubeF7 v1.14 with CubeMX 5.01

elgarbe, 2020-11-12:
this issue is not present on CubeF7 v1.15 with CubeMX 5.20

ASELSTM, 2020-11-13:
Hi @elgarbe,

Thank you for this report.

We tried to reproduce the issue by generating the code with **CubeMx V6.0.1** and the latest release of **STM32F7CubeFW (V1.16.0)** and we couldn't notice the problem. 

For a further understanding of the problem, may i ask you some questions : 

1. Could you please try to generate the project using the latest version of **STM32CubeMX(V6.0.1)**  and the **CubeFW V1.14.0** or **CubeFW V1.15.0** and check if you still have the same issues.

2. Would you please share the ioc files obtained by **CubeMXV5.0.1** and  **CubeMXV5.0.2**.

With regards,

elgarbe, 2020-11-14:
Hi, which board do you test? how do you try to reproduce this?
STM32CubeMX 6.0.1 can't generate a project that doesn't compile using CubeFW v1.14.0 so I can't test
STM32CubeMX 6.0.1 and CubeFW v1.15.0 doesn't present the issue I'm reporting

I already delete CubeMXv5.01 so I can't provide the .ioc file with that version.

My workaround right now is to use CubeMX 6.0.1 with CubeFW v1.15.0. But you should confirm this bug or share a .ioc/SystemWorkbench project for stm32f722re without this issue.

ASELSTM, 2021-01-15:
Hi  @elgarbe,

We do apologize for the delay we take to answer sometimes.

We used the IOC file you have shared to generate the project using the same CubeFW version (STM32Cube FW_F7 V1.16.0) and tested on an STM32F722RE and an STM32F767ZI Nucleo board.

We have tested on 2 different toolchains ( EWARM and STM32CubeIDE ) and unfortunately we couldn't reproduce the issue. 
 
With regards,

ASELSTM, 2021-02-02:
Hi @elgarbe,

Please allow me to close this thread as no activity has been registered. You can reopen it at any time if you have details to provide us to help you solve this issue.
Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-02-02 by ASELSTM