STMicroelectronics/STM32CubeF7 #34: Used NULL value for uint32_t osMessageQueuePut() parameter in STM32_USB_Host_Library

stsymbaliuk, 2020-11-18:
STM32CubeMX Version 6.1.0
STM32CubeF7 Firmware Package V1.16.0 / 14-February-2020

For next and similar `osMessageQueuePut()` call:
https://github.com/STMicroelectronics/STM32CubeF7/blob/08376dce1b404687e4a86b73077f396bccfc9cb5/Middlewares/ST/STM32_USB_Host_Library/Class/MSC/Src/usbh_msc_bot.c#L220-L224

My `osCMSIS` from cmsis_os.h
```c
#define osCMSIS               0x20001U  ///< API version (main[31:16].sub[15:0])
```

Get lot of warnings with `-Wint-conversion` compiler option:
```
D:/Work/prj_name/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_core.c:443:64: warning: passing argument 4 of 'osMessageQueuePut' makes integer from pointer without a cast [-Wint-conversion]
   (void)osMessageQueuePut(phost->os_event, &phost->os_msg, 0U, NULL);
                                                                ^~~~
In file included from ../../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2/cmsis_os.h:155:0,
                 from ../../USB_HOST/Target/usbh_conf.h:98,
                 from ../../Middlewares/ST/STM32_USB_Host_Library/Core/Inc/usbh_core.h:28,
                 from D:/Work/LL/landru/landru-mainboard-boot/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_core.c:22:
../../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS_V2/cmsis_os2.h:689:12: note: expected 'uint32_t {aka long unsigned int}' but argument is of type 'void *'
 osStatus_t osMessageQueuePut (osMessageQueueId_t mq_id, const void *msg_ptr, uint8_t msg_prio, uint32_t timeout);
            ^~~~~~~~~~~~~~~~~
```

From `cmsis_os2.h` file `osMessageQueuePut()` declaration is:
```c
osStatus_t osMessageQueuePut (osMessageQueueId_t mq_id, const void *msg_ptr, uint8_t msg_prio, uint32_t timeout);
```

And from `stddef.h` file:
```c
#define NULL ((void *)0)
```

Possible fix is to replace all
```c
(void)osMessageQueuePut(phost->os_event, &phost->os_msg, 0U, NULL);
```
with
```c
(void)osMessageQueuePut(phost->os_event, &phost->os_msg, 0U, 0);
```

Affeccted files:
- Middlewares/ST/STM32_USB_Host_Library/Class/MSC/Src/usbh_msc.c
- Middlewares/ST/STM32_USB_Host_Library/Class/MSC/Src/usbh_msc_bot.c
- Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_core.c
- Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq

ALABSTM, 2021-01-19:
Hi @stsymbaliuk,

Thank you for your report and the fix proposal. The issue has been confirmed by our development teams. A fix will be made available in a future version. Thank you again.

With regards,

ALABSTM, 2021-01-19:
ST Internal Reference: 99700

FBergemann, 2021-06-20:
I hit the same problem. 
As long as this fix is not releases: is it required to manually change NULL -> 0? Or does it also work without this modification?

stsymbaliuk, 2021-06-21:
@FBergemann code works without replacing. But to avoid compiler warning that I've placed in issue description I've replaced it locally on my PC.

ALABSTM, 2021-06-21:
Hi @stsymbaliuk,

I hope you are doing well. This point you raised has been fixed in the frame of **v3.4.0** of the USB Host Library found in the [stm32_mw_usb_host](https://github.com/STMicroelectronics/stm32_mw_usb_host) repository. Thank you again for your report and proposal.

With regards,

FBergemann, 2021-06-26:
Hi @stsymbaliuk 
i filed another issue here: https://github.com/STMicroelectronics/STM32CubeF7/issues/46
Maybe you also will hit that.
regards,
Frank

ASELSTM, 2022-07-08:
Hi @stsymbaliuk,

Thank you for your contribution. This issue has been fixed in the frame of version v1.17.0 of the STM32CubeF7. Please allow me then to close this thread.

Thank you again for your contribution.

With regards,

closed 2022-07-08 by ASELSTM