STMicroelectronics/STM32CubeF7 #35: Wrong prescaler value in HAL_InitTick() when Timer clocks less than 1MHz

stsymbaliuk, 2020-11-27:
TIM7 selected as Timebase Source
HCLK in Clock config set to 1 MHz
APB1 Prescaler is 4
APB1 Timer clocks: 0.5 MHz

Observed behavior looks like `uwTick` does not change but in fact due to very high prescaler it counts very slow.

File `stm32f7xx_hal_timebase_tim.c`, `HAL_InitTick()` function
```c
/* Compute TIM7 clock */
uwTimclock = 2*HAL_RCC_GetPCLK1Freq(); // 2 * 250000 = 500000
/* Compute the prescaler value to have TIM7 counter clock equal to 1MHz */
uwPrescalerValue = (uint32_t) ((uwTimclock / 1000000U) - 1U); // (500000U / 1000000U) - 1U = 0 - 1U = 4294967295 (max uint32_t value)
```

So for `uwTimclock` less than 1 million `uwPrescalerValue` must be 0 and not `4294967295`.

Possible fix is to replace string
```c
uwPrescalerValue = (uint32_t) ((uwTimclock / 1000000U) - 1U);
```

with
```c
uint32_t mhz = (uwTimclock / 1000000U);

if (mhz > 1)
{
  uwPrescalerValue = mhz - 1;
}
else
{
  uwPrescalerValue = 0;
}
```

STM32CubeMX Version 6.1.0
STM32CubeF7 Firmware Package V1.16.0 / 14-February-2020

[image]

RKOUSTM, 2020-12-02:
Hi @stsymbaliuk,

Thank you for your contribution.  In order to allow a better analysis of this problem, could you please give us your .ioc file that you are used to reproduce this issue. Thank you again for your contribution.

With regards,

stsymbaliuk, 2020-12-03:
Hi @RKOUSTM 
Here is my project and .ioc file for STM32F769I-DISCO board.
MCU: STM32F69NIH
STM32CubeMX Version 6.1.0
STM32CubeF7 Firmware Package V1.16.0 / 14-February-2020

TIM7 selected as Timebase Source
HCLK in Clock config set to 1 MHz
APB1 Prescaler is 4
APB1 Timer clocks: 0.5 MHz

In main() I have a loop which must toggle LEDs on board every 1000 ms.
I've already added fixes described in my previous message to `stm32f7xx_hal_timebase_tim.c` to make it work.
If you generate project from CubeMX again changes in `stm32f7xx_hal_timebase_tim.c` will be overwritten end problem appear. HAL_Delay() will be broken and LEDs will be on for a very long time.

[stm32f769i-discovery_issue_35.zip](https://github.com/STMicroelectronics/STM32CubeF7/files/5635642/stm32f769i-discovery_issue_35.zip)

RKOUSTM, 2020-12-18:
Hi @stsymbaliuk,

Thank you for your contribution. After investigation, the point you reported is related to **CubeMX** tool . According to the following .ioc file, an update on the system clock configuration is proposed to solve your reported problem.

[image]

In general terms, we don't treat aspects related to **CubeMX** tool at GitHub level, they are rather treated at ST Community level.
Now, as this issue is not directly related to software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to ecosystem (**CubeMX** tool), please allow me to close it.

Thank you for your comprehension and thank you again for your contribution. 

With regards,

closed 2020-12-22 by RKOUSTM