STMicroelectronics/STM32CubeF7 #36: Update FatFs (ff.h) to latest version

ladislas, 2020-12-10:
**Describe the set-up**

DISCO_F769NI + Mbed 0S

**Describe the bug**

When trying to port examples using FatFs (ff.h) from CubF7 to mbed, a lot of issues arise because of the difference of versions:

- CubeF7 - R0.12c - https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Middlewares/Third_Party/FatFs/src/ff.h#L2

- Mbed SO - R0.13a - https://github.com/ARMmbed/mbed-os/blob/master/storage/filesystem/fat/ChaN/ff.cpp#L2

ST's R0.12c is more than 3 years old... (March 04, 2017) 

It might be time to upgrade to the latest version R0.14a (December 5, 2020) and be able to benefit from 3 years of bug fixes and improvements.

**How To Reproduce**

Try porting any examples using FatFs to Mbed OS

**Additional context**

Update to latest version + update examples

**Screenshots**

n/a

RKOUSTM, 2020-12-15:
Hi @ladislas,

Thank you for your contribution. We have reported your request to our technical team, we still cannot share a date for the moment for the integration of FatFs version R0.13a. 

Please, allow me to close this issue. Thank you for your comprehension.

With regards,

ladislas, 2020-12-15:
Thanks @RKOUSTM for the feedback.

There's an open PR on mbed-os to update FatFs to R0.14a

https://github.com/ARMmbed/mbed-os/pull/14028

It would be great to update to the same version (instead of R0.13a)

closed 2020-12-16 by RKOUSTM