STMicroelectronics/STM32CubeF7 #37: Error in usbd_conf_template.c line 178. 'ep_addr' undeclared

emielsteerneman, 2021-01-15:
https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Middlewares/ST/STM32_USB_Device_Library/Core/Src/usbd_conf_template.c 

In this file, on line 178, the variable 'ep_addr' is used but undeclared.

ALABSTM, 2021-01-22:
Hi @emielsteerneman,

Thank you for having reported this point. It will be logged into our internal database to be fixed. Indeed, the parameter should be `dev_addr` instead of `ep_addr`. Thank you again.

With regards,

ALABSTM, 2021-01-22:
ST Internal Reference: 99960

ALABSTM, 2021-06-21:
Hi @emielsteerneman,

I hope you are fine. The point you have raised has been updated in the frame of **v2.8.0** of the USB Device Library found in the [stm32_mw_usb_device](https://github.com/STMicroelectronics/stm32_mw_usb_device) repository. Thank you again for your report.

With regards,

ALABSTM, 2021-11-15:
Hi @emielsteerneman,

Back to you about this point. Unfortunately, there is no plan to integrate version **v2.8.0** of the USB Device library. However, you can get the latest updates from the repository indicated above as all versions are backward compatible.

Please allow me to close this issue. Thank you for your comprehension and thank you again for having reported the point.

With regards,

closed 2021-11-15 by ALABSTM

emielsteerneman, 2022-01-10:
@ALABSTM I do not understand. There is a bug in the code. It has been fixed in stm32_mw_usb_device six months ago, but is still present in the master branch to this day.

ALABSTM, 2022-01-11:
Hi @emielsteerneman,

As mentioned before, unfortunately, our development teams have **no plan** to integrate the new USB Device library version to the STM32CubeF7 firmware, this one still supporting a previous library version. This is why this issue has been closed with the label "wontfix".

With regards,