STMicroelectronics/STM32CubeF7 #38: STM32F767ZI-Nucleo/LwIP_HTTP_Server_Netconn_RTOS project not compiling when ported on STM32CubeIDE

rahmanih, 2021-01-23:
**Describe the set-up**
 * Board : STM32F767ZI_Nucleo
 * STM32CubeIDE 1.5.1

**Describe the bug**
*  Build the project  using the STM32CubeIDE genreates the following error:
```
make all 
arm-none-eabi-gcc "C:/work/releases/STM32Cube_FW_F7_V1.16.0/Middlewares/Third_Party/LwIP/src/apps/http/fs.c" -mcpu=cortex-m7 -std=gnu11 -g3 -DSTM32F767xx -DUSE_HAL_DRIVER -DUSE_STM32F7XX_NUCLEO_144 -c -I../../../Inc -I../../../Src -I../../../../../../../../Drivers/CMSIS/Device/ST/STM32F7xx/Include -I../../../../../../../../Drivers/CMSIS/Include -I../../../../../../../../Drivers/STM32F7xx_HAL_Driver/Inc -I../../../../../../../../Drivers/BSP/STM32F7xx_Nucleo_144 -I../../../../../../../../Drivers/BSP/Components/Common -I../../../../../../../../Drivers/BSP/Components -I"../../../../../../../../Middlewares/Third_Party/LwIP/src/include/      " -I../../../../../../../../Middlewares/Third_Party/LwIP/system -I../../../../../../../../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM7/r0p1 -I../../../../../../../../Middlewares/Third_Party/FreeRTOS/Source -I../../../../../../../../Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS -I../../../../../../../../Middlewares/Third_Party/FreeRTOS/Source/include -O1 -ffunction-sections -Wall -fstack-usage -MMD -MP -MF"Middlewares/LwIP/apps/http/fs.d" -MT"Middlewares/LwIP/apps/http/fs.o" --specs=nano.specs -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -o "Middlewares/LwIP/apps/http/fs.o"
C:/work/releases/STM32Cube_FW_F7_V1.16.0/Middlewares/Third_Party/LwIP/src/apps/http/fs.c:33:10: fatal error: lwip/apps/httpd_opts.h: No such file or directory
   33 | #include "lwip/apps/httpd_opts.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Middlewares/LwIP/apps/http/subdir.mk:21: Middlewares/LwIP/apps/http/fs.o] Error 1
```
**How To Reproduce**
* Import the project _Projects/STM32F767ZI-Nucleo/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/SW4STM32_ using the STM32CubeIDE
* build the project

** Root Cause **
[this include path](https://github.com/STMicroelectronics/STM32CubeF7/blob/08376dce1b404687e4a86b73077f396bccfc9cb5/Projects/STM32F767ZI-Nucleo/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/SW4STM32/STM32F767ZI_Nucleo/.cproject#L44) in the poject is wrong as it contains trailing white spaces

regards
Haithem.

RKOUSTM, 2021-01-25:
Hi @rahmanih,

Thank you for reporting this report. You are absolutely right about this point. An internal bug tracker has been already opened. The fix will be made available in the frame of a future release.

Thank you once more for your contribution.

With regards,

RKOUSTM, 2021-01-25:
ST Internal Reference: 93485

RKOUSTM, 2021-03-03:
Hi @rahmanih,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.16.1 of the STM32CubeF7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-03 by RKOUSTM