STMicroelectronics/STM32CubeF7 #39: Missing support for NUCLEO-F756ZG

mbates14, 2021-02-02:
I noticed that this particular board is not in the board support. 

any reason as to why?

ASELSTM, 2021-02-05:
Hi @mbates14,

First we would like to thank you for your contribution. Actually, only a selection of boards is supported in the **"Board Selector"** list. Indeed, we cannot support the totality of the boards deployed by STMicroelectronics. However, you can find the MCU of this board in the **"MCU/MPU selector"** as shown below. 

[image]

Hopefully this would help you. 

With regards,

ASELSTM, 2021-02-16:
Hi @mbates14,

Please allow me to close this thread.  You can reopen it at any time if you have details to provide us to help you.
Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-02-16 by ASELSTM

mbates14, 2021-02-23:
Not a solution at all. 

Sure, I could pick the exact CPU, but now you are missing the complete board support package. So you have to resetup all the GPIOs, set up the whole clocking system, and everything from scratch at that point instead of just picking the correct board. Nobody has time for that in a RAD environment.

CharlesMillerM4, 2021-11-12:
@mbates14, I completely agree.  Currently, the F746 Nucleo version is unavailable from several distributors (as of 11/12/2021), but the F756 version is.

To create my project, I had to create a "blank" Nucleo-F746 Cube project, then create a blank 756 MCU project, then configure the latter's clocks, pins, etc. to match the 746 board.  It would not take the ST Cube app writers to add Nucleo STM32F756ZG support, since the '46 and '56 chips are identical, save the Crypto block.

"...Can't add support for all boards..."   ?? C'mon, ST, you can do better than this!
(I also note that ALL the other F7 boards--Nucleo and Discovery--have project generation!)

CharlesMillerM4, 2021-12-07:
As of CubeMX version 6.4.0, Nucleo-F756ZG support is available.  Congratulations, STMicro, and thank you!