STMicroelectronics/STM32CubeF7 #40: lwIP Static and FreeRTOS does not work.

mbates14, 2021-02-09:
So where do I start.... 

if I create a new project from scratch in the STM32CubeIDE, using the latest 1.16 firmware release. 

I pick my LwIP middleware, including FreeRTOS. 

the system auto-builds the code. But if I use a Static IP instead of DHCP, I cannot ping the board. with DHCP enabled, I can ping the board no problem. 

This is bone stock without any additional code modifications. 

I tried using the MQTT library built into LwIP on DHCP, but the connect keeps failing with return code -4, cannot find an explanation on the internet. 

But, with debuggin information turned on, and redirect stdout to UART, I get a weird crash with Static IP configuration like this: 

MEMP_SYS_TIMEOUT empty error. ASSERT failed thing. (paraphrasing here) 

This seems like an issue, not sure what. Thoughts?

RKOUSTM, 2021-02-23:
Hi @mbates14 ,

Thank you for your contribution. The problem seems to be related to migration to the CubeIDE. However, this public organization on GitHub is intended to promote co-development and issues posted here shall exclusively be related to the software published within the different repositories.
For any other issue (e.g. related to documentation, to hardware, to ecosystem, to user application, etc.), you shall address your questions to the [ST Community](https://community.st.com/s/topiccatalog). There you shall look for the adequate topic and submit your question or remark.

You can use the IDE proposed by ST (IAR, MDK-ARM and SW4STM32) for middleware application on STM32F7 board all of them are working well.

Please allow me to close this issue now. Thank you again for you contribution.

With regards,

closed 2021-02-23 by RKOUSTM

mbates14, 2021-02-23:
This is still an issue. I had to use a workaround by starting the lwIP stack in a thread instead of main();