STMicroelectronics/STM32CubeF7 #42: assertion "REENT malloc succeeded" failed

muskec06, 2021-04-05:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.

**Describe the bug**
A clear and concise description of what the bug is.

**How To Reproduce**
1. Indicate the global behavior of your application project.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).

3. The use case that generates the problem.

4. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

muskec06, 2021-04-05:
I am using stm32CubeIde 1.6.1 and latest drivers. I am using Lwip on stm32f769-disco kit. Before when I use previous drivers I couldn't get a message (In UART console) as below

assertion "REENT malloc succeeded" failed: file "/build/gnu-tools-for-stm32_9-2020-q2-update.2020100              1-1621/src/newlib/newlib/libc/stdlib/rand.c", line 78

ASELSTM, 2021-04-08:
Hi @muskec06,

Thank you for this report. If the issue is systematically reproducible on your side, would you please give us more details about how to reproduce the issue and if possible share the project so that we can reproduce the same behavior.

With regards,

Malangcow, 2021-05-20:
I am also experiancing this issue.

**Describe the set-up**

The board (either ST RPN reference or your custom board).
STM32F429I-DISCO

IDE or at least the compiler and its version.
STM32CubeIDE 1.6.1
CubeMX - Software Pack - X-CUBE-TOUCHGFX.4.16.1

**Describe the bug**
A clear and concise description of what the bug is.

assertion fail when rand() is called.

Following is the assertion message printed in the console.

```
: Assertion: 'REENT malloc succeeded' failed. /build/gnu-tools-for-stm32_9-2020-q2-update.20201001-1621/src/newlib/newlib/libc/stdlib/rand.c:78
```

**How To Reproduce**

1. Generate project using TouchGFX Designer.

2. UI Template used from the designer was TouchGFX Demo 3(Since the display resolution of the board is 320 by 240)

3. From the main screen add a widget(button with label)

4. Configure interaction of the button. Trigger: 'Button is clicked', Action: call virtual function 'button1Clicked'

5. STM32CubeIDE: Import generated project.

6. Run CubeMX from the CubeIDE. Configure peripherals, onboard LED.

7. Implement function 'button1Clicked'
   ```cpp
   void MainView::button1Clicked()
   {
   	HAL_GPIO_TogglePin(LED_LD3_GPIO_Port, LED_LD3_Pin);
   	HAL_GPIO_TogglePin(LED_LD4_GPIO_Port, LED_LD4_Pin);
     
     // Set Random Color
     auto randColor = rand();
     dlogln("%s: rand = %08Xh", __PRETTY_FUNCTION__, randColor);
     uint8_t rgb_r = randColor & 0xFF;
     uint8_t rgb_g = (randColor >> 8) & 0xFF;
     uint8_t rgb_b = (randColor >> 16) & 0xFF;
     buttonWithLabel1.setLabelColor(touchgfx::Color::getColorFrom24BitRGB(rgb_r, rgb_g, rgb_b));
     buttonWithLabel1.invalidate();
   }
   ```
   
8. Build project. Program firmware binary.

9. Run the board.

10. Click the button widget.

11. Assertion fail message is printed and TouchGFX task halts.

**Screenshots**

[image]

ASELSTM, 2021-07-07:
Hi,

Please allow me to close this thread as a duplicate of [this ](STMicroelectronics/STM32CubeF4#75)issue.

With regards,

closed 2021-07-07 by ASELSTM