STMicroelectronics/STM32CubeF7 #45: HAL MMC card initialization always fails

nickstoughton, 2021-06-17:
**Setup**
 * Nucleo-F746ZG board, but present for at least all F7xx processors (and probably many others)
 * Found using gcc 10.1

**Problem**
Any attempt to call `MMC_InitCard()` will fail

**How To Reproduce**
1.Attach an eMMC card (1-bit, 4-bit or 8-bit data paths, it will not make a difference) to a`SDMMC` port and attempt to use it. (NOTE, eMMC cards use the JEDEC MMC protocol, and not the SD Card protocol)

2.Attempt to initialize it. This call will fail under all circumstances.

**Patch**
The full patch follows, but note that this does change the underlying low level API. If necessary, I can reformulate it to leave that original (but badly named) function, and rename the new function.

From: Nick Stoughton <nstoughton@logitech.com>
Date: Wed, 16 Jun 2021 13:59:57 -0700
Subject: [PATCH] stm32f7xx HAL and LL SD driver fixes

Differentiate between the MMC CMD3 (SET_RELATIVE_ADDR) and the SDIO CMD3
(SEND_RELATIVE_ADDR).

The hal_mmc driver had two bugs:
1. It uses the SEND_RELATIVE_ADDR command, which relies on the card
assigning itself an RCA and returning that in R6. However, the JEDEC MMC
standard has the host assign the RCA and send it as part of CMD3. The
result of CMD3 is in R1, not R6.
2. After the RCA has been assigned, the card must be selected before any
command that requires a data transfer (e.g. CMD8 to get the extended
CSD). If the card is not selected, any command that involves a data
transfer will timeout and no data will be transferred.

This proposal provides two alternative versions of the low level CMD3
function. `SDMMC_CmdSetRelAdd` (old name but new parameters) supports
the MMC protocol, while `SDMMC_CmdSendRelAdd` is provided to support the
SD protocol. However, this changes the published Low Level
specification.

```
 .../drivers/include/stm32f7xx_ll_sdmmc.h      |  3 ++-
 .../stm32f7xx/drivers/src/stm32f7xx_hal_mmc.c | 14 +++++-----
 .../stm32f7xx/drivers/src/stm32f7xx_hal_sd.c  |  2 +-
 .../drivers/src/stm32f7xx_ll_sdmmc.c          | 27 ++++++++++++++++++-
 4 files changed, 36 insertions(+), 10 deletions(-)

diff --git a/stm32cube/stm32f7xx/drivers/include/stm32f7xx_ll_sdmmc.h b/stm32cube/stm32f7xx/drivers/include/stm32f7xx_ll_sdmmc.h
index 42d83107..b8477ce0 100644
--- a/stm32cube/stm32f7xx/drivers/include/stm32f7xx_ll_sdmmc.h
+++ b/stm32cube/stm32f7xx/drivers/include/stm32f7xx_ll_sdmmc.h
@@ -987,7 +987,8 @@ uint32_t SDMMC_CmdSendSCR(SDMMC_TypeDef *SDMMCx);
 uint32_t SDMMC_CmdSendCID(SDMMC_TypeDef *SDMMCx);
 uint32_t SDMMC_CmdSendCSD(SDMMC_TypeDef *SDMMCx, uint32_t Argument);
 uint32_t SDMMC_CmdSendEXTCSD(SDMMC_TypeDef *SDMMCx, uint32_t Argument);
-uint32_t SDMMC_CmdSetRelAdd(SDMMC_TypeDef *SDMMCx, uint16_t *pRCA);
+uint32_t SDMMC_CmdSetRelAdd(SDMMC_TypeDef *SDMMCx, uint16_t RCA);
+uint32_t SDMMC_CmdSendRelAdd(SDMMC_TypeDef *SDMMCx, uint16_t *pRCA);
 uint32_t SDMMC_CmdSendStatus(SDMMC_TypeDef *SDMMCx, uint32_t Argument);
 uint32_t SDMMC_CmdStatusRegister(SDMMC_TypeDef *SDMMCx);
 uint32_t SDMMC_CmdOpCondition(SDMMC_TypeDef *SDMMCx, uint32_t Argument);
diff --git a/stm32cube/stm32f7xx/drivers/src/stm32f7xx_hal_mmc.c b/stm32cube/stm32f7xx/drivers/src/stm32f7xx_hal_mmc.c
index 619665c8..c3db412b 100644
--- a/stm32cube/stm32f7xx/drivers/src/stm32f7xx_hal_mmc.c
+++ b/stm32cube/stm32f7xx/drivers/src/stm32f7xx_hal_mmc.c
@@ -2600,7 +2600,7 @@ static uint32_t MMC_InitCard(MMC_HandleTypeDef *hmmc)
 
   /* Send CMD3 SET_REL_ADDR with argument 0 */
   /* MMC Card publishes its RCA. */
-  errorstate = SDMMC_CmdSetRelAdd(hmmc->Instance, &mmc_rca);
+  errorstate = SDMMC_CmdSetRelAdd(hmmc->Instance, mmc_rca);
   if(errorstate != HAL_MMC_ERROR_NONE)
   {
     return errorstate;
@@ -2627,12 +2627,6 @@ static uint32_t MMC_InitCard(MMC_HandleTypeDef *hmmc)
   /* Get the Card Class */
   hmmc->MmcCard.Class = (SDMMC_GetResponse(hmmc->Instance, SDMMC_RESP2) >> 20U);
 
-  /* Get CSD parameters */
-  if (HAL_MMC_GetCardCSD(hmmc, &CSD) != HAL_OK)
-  {
-    return hmmc->ErrorCode;
-  }
-
   /* Select the Card */
   errorstate = SDMMC_CmdSelDesel(hmmc->Instance, (uint32_t)(((uint32_t)hmmc->MmcCard.RelCardAdd) << 16U));
   if(errorstate != HAL_MMC_ERROR_NONE)
@@ -2640,6 +2634,12 @@ static uint32_t MMC_InitCard(MMC_HandleTypeDef *hmmc)
     return errorstate;
   }
 
+  /* Get CSD parameters */
+  if (HAL_MMC_GetCardCSD(hmmc, &CSD) != HAL_OK)
+  {
+    return hmmc->ErrorCode;
+  }
+
   /* Configure SDMMC peripheral interface */
   (void)SDMMC_Init(hmmc->Instance, hmmc->Init);
 
diff --git a/stm32cube/stm32f7xx/drivers/src/stm32f7xx_hal_sd.c b/stm32cube/stm32f7xx/drivers/src/stm32f7xx_hal_sd.c
index 3886cf37..3c1d3bf6 100644
--- a/stm32cube/stm32f7xx/drivers/src/stm32f7xx_hal_sd.c
+++ b/stm32cube/stm32f7xx/drivers/src/stm32f7xx_hal_sd.c
@@ -2647,7 +2647,7 @@ static uint32_t SD_InitCard(SD_HandleTypeDef *hsd)
   {
     /* Send CMD3 SET_REL_ADDR with argument 0 */
     /* SD Card publishes its RCA. */
-    errorstate = SDMMC_CmdSetRelAdd(hsd->Instance, &sd_rca);
+    errorstate = SDMMC_CmdSendRelAdd(hsd->Instance, &sd_rca);
     if(errorstate != HAL_SD_ERROR_NONE)
     {
       return errorstate;
diff --git a/stm32cube/stm32f7xx/drivers/src/stm32f7xx_ll_sdmmc.c b/stm32cube/stm32f7xx/drivers/src/stm32f7xx_ll_sdmmc.c
index 042c4f93..0e1d85a2 100644
--- a/stm32cube/stm32f7xx/drivers/src/stm32f7xx_ll_sdmmc.c
+++ b/stm32cube/stm32f7xx/drivers/src/stm32f7xx_ll_sdmmc.c
@@ -1001,13 +1001,38 @@ uint32_t SDMMC_CmdSendCSD(SDMMC_TypeDef *SDMMCx, uint32_t Argument)
   return errorstate;
 }
 
+/**
+  * @brief  Send the Set CSD command and check the response.
+  * @param  SDMMCx: Pointer to SDMMC register base
+  * @param  RCA: Card RCA
+  * @retval HAL status
+  */
+uint32_t SDMMC_CmdSetRelAdd(SDMMC_TypeDef *SDMMCx, uint16_t RCA)
+{
+  SDMMC_CmdInitTypeDef  sdmmc_cmdinit;
+  uint32_t errorstate;
+
+  /* Send CMD3 SD_CMD_SET_REL_ADDR */
+  sdmmc_cmdinit.Argument         = RCA << 16;
+  sdmmc_cmdinit.CmdIndex         = SDMMC_CMD_SET_REL_ADDR;
+  sdmmc_cmdinit.Response         = SDMMC_RESPONSE_SHORT;
+  sdmmc_cmdinit.WaitForInterrupt = SDMMC_WAIT_NO;
+  sdmmc_cmdinit.CPSM             = SDMMC_CPSM_ENABLE;
+  (void)SDMMC_SendCommand(SDMMCx, &sdmmc_cmdinit);
+
+  /* Check for error conditions */
+  errorstate = SDMMC_GetCmdResp1(SDMMCx, SDMMC_CMD_SET_REL_ADDR, SDMMC_CMDTIMEOUT);
+
+  return errorstate;
+}
+
 /**
   * @brief  Send the Send CSD command and check the response.
   * @param  SDMMCx: Pointer to SDMMC register base
   * @param  pRCA: Card RCA
   * @retval HAL status
   */
-uint32_t SDMMC_CmdSetRelAdd(SDMMC_TypeDef *SDMMCx, uint16_t *pRCA)
+uint32_t SDMMC_CmdSendRelAdd(SDMMC_TypeDef *SDMMCx, uint16_t *pRCA)
 {
   SDMMC_CmdInitTypeDef  sdmmc_cmdinit;
   uint32_t errorstate;
```

ASELSTM, 2021-07-06:
Hi @nickstoughton,

Thank you for your contribution and for the proposed fix. We are actually aware about the point you raised out and are working on fixing it. The fix will be made available in the frame of a future release. 
We cannot share a date for the moment. So stay tuned and thank you once more for your contribution and for your patience.

With regards

ASELSTM, 2021-07-06:
ST Internal Reference: 83264

ASELSTM, 2022-01-14:
Hi @nickstoughton,

This issue has been fixed in the frame of version **v1.16.2** of the STM32CubeF7 published some time ago. Please allow me then to close it.

With regards,

closed 2022-01-14 by ASELSTM