STMicroelectronics/STM32CubeF7 #46: void MX_DriverVbusFS(uint8_t state) - wrong state evaluation?

FBergemann, 2021-06-26:
Can you please have a look at plugins/com.st.stm32cube.common.mx_6.2.1.202103241236/db/templates/usbh_ 
platform_c.ftl ? And tell, if the 'state' evaluation is wrong here?

This is how it ended up in my USB_HOST/Target/usbh_platform.c:

`/**`
`  * @brief  Drive VBUS.`
`  * @param  state : VBUS state`
`  *          This parameter can be one of the these values:`
`  *           - 1 : VBUS Active`
`  *           - 0 : VBUS Inactive`
`  */`
`void MX_DriverVbusFS(uint8_t state)`
`{`
`  uint8_t data = state;`
`  /* USER CODE BEGIN PREPARE_GPIO_DATA_VBUS_FS */`
`  if(state == 0)`
`  {`
`    /* Drive high Charge pump */`
`    data = GPIO_PIN_SET;`
`  }`
`  else`
`  {`
`    /* Drive low Charge pump */`
`    data = GPIO_PIN_RESET;`
`  }`
`  /* USER CODE END PREPARE_GPIO_DATA_VBUS_FS */`
`  HAL_GPIO_WritePin(GPIOG,GPIO_PIN_6,(GPIO_PinState)data);`
`}`

FBergemann, 2021-07-04:
Please let me adopt and slightly modify comment from other issue #47 here - because it's about 2 issue in the same area:

BTW
I didn't find any reference project for "USB_HOST / Mass Storage Host Class" that is using the Device Configuration Tool (CubeMX). All examples/references i was given, didn't use *.ioc file. **It would be good to have examples, that are based on the Device Configuration Tool (CubeMX). Then those problems would have shown up already, before.**
Note, that i also filed another issue [Combination of USB_HOST, FATFS, and FreeRTOS crashes for too small stack size](https://github.com/STMicroelectronics/STM32CubeF7/issues/47)  for this. And currently i am struggling with a 3rd problems, which is about wrong intialization of PA10 (USB_OTS_FS_ID) (i will create another separate issue for that).

RKOUSTM, 2021-08-13:
Hi @FBergemann,

Thank you for your contribution. You are absolutely right about this point. Actually, the point you raised has already been dealt and fixed internally. The fix will be made available in the frame of a future release of CubeMX.

You can refer to [this](https://github.com/STMicroelectronics/STM32CubeF7/tree/master/Projects/STM32F767ZI-Nucleo/Applications/USB_Host/MSC_Standalone) USB_Host-MSC_Standalone application where the VBUS state of the USBH_LL_DriverVBUS() function in the file usbh_conf.c is correctly listed.

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to our ecosystem (namely the Cube MX tool), please allow me to close it.

Thank you for your contribution. We are looking forward to reading from you again.

With regards,

RKOUSTM, 2021-08-13:
ST Internal Reference: 110084.

RKOUSTM, 2021-08-13:
ST Internal Reference: 110620.

closed 2021-08-13 by RKOUSTM