STMicroelectronics/STM32CubeF7 #48: CubeMX: USB_OTG_FS / USB_HOST does not initialize PA10

FBergemann, 2021-07-04:
I am trying to set up USB_OTG_FS / USB_HOST for NUCLEO-F767ZI - with the "Device Configuration Tool" (CubeMX).

For creating a new STM32 Project for NUCLEO-F767ZI, in the beginning, i
- did not initialize all peripherals
- i cleared all PINs in the Device Configuration Tool (CubeMX) view 

Then next:
-  edited "System Core / RCC" and enabled High Speed Clock (HSE): Crystal/Ceramic Resonator
-  enabled "Middleware / FreeRTOS", Interface: CMSIS_V1 or CMSIS_V2 (it doesn't make a difference, which is select)
-  enabled "Connectivity / USB_OTG_FS", Mode: Host_Only, [x] Activate_VBUS
-  enabled "Middleware / USB_HOST", Class for FS IP: Mass Storage Host Class
-  saved *.ioc file

The resulting usb_conf.c, does **not** have any initialization for PA10 (resp. GPIO_PIN_10 resp. USB_OTS_FS_ID).

But all reference projects, i'd been given initialize like this:
`    /* Configure ID pin */`
`    GPIO_InitStruct.Pin = GPIO_PIN_10;`
`    GPIO_InitStruct.Mode = GPIO_MODE_AF_OD;`
`    GPIO_InitStruct.Pull = GPIO_PULLUP;`
`    GPIO_InitStruct.Alternate = GPIO_AF10_OTG_FS;`
`    HAL_GPIO_Init(GPIOA, &GPIO_InitStruct);`

Those reference projects are:
- Projects\STM32F767ZI-Nucleo\Applications\USB_Host\MSC_Standalone
- Projects\STM32F769I_EVAL\Applications\USB_Host\MSC_RTOS 
- Projects\STM32F769I_EVAL\Applications\USB_Host\MSC_Standalone

Is the initialization of PA10 missing?

FBergemann, 2021-07-04:
BTW
Note, that i also filed more issues in the same context: 
- [void MX_DriverVbusFS(uint8_t state) - wrong state evaluation? for this](https://github.com/STMicroelectronics/STM32CubeF7/issues/47) 
- [Combination of USB_HOST, FATFS, and FreeRTOS crashes for too small stack size](https://github.com/STMicroelectronics/STM32CubeF7/issues/47)

I didn't find any reference project for "USB_HOST / Mass Storage Host Class" that is using the Device Configuration Tool (CubeMX).
All examples/references i was given, didn't use *.ioc file.
**It would be good to have examples, that are based on the Device Configuration Tool (CubeMX). 
Then those problems would have shown up already, before.**

RKOUSTM, 2021-07-14:
Hi @FBergemann,

This duplicated issue will remain closed. A further detail of your reported point is answered in Issue #47.

With regards,

RKOUSTM, 2021-07-14:
Duplicate of #47.

closed 2021-07-14 by RKOUSTM