STMicroelectronics/STM32CubeF7 #49: stm32 lwip sample projects have problem at cable unplug mcu start

mohammadpoorkhalili, 2021-08-02:
hi . when i using every sample from lwip with every st official evaluation board , when mcu start without cable connected , after plug cable device connot connect to network.

mohammadpoorkhalili, 2021-08-23:
hi this problem when freertos is not enabled, can be solve by editing MX_LWIP_Init function.
but when i enabled freertos can't find way for solve this problem.

ASELSTM, 2022-01-14:
Hi @mohammadpoorkhalili,

This is actually a normal behavior. Indeed, after timeout connection, the device gets static address.
You might then change the application behavior to use only dynamic address or to use only static address or to switch differently from dynamic to static.

As this issue is not valid, please allow me then to close this thread.

With regards,

closed 2022-01-14 by ASELSTM