STMicroelectronics/STM32CubeF7 #50: stm32f7xx_hal_rcc.c has a broken PLLN check

InconsolableCellist, 2021-08-06:
See: https://github.com/STMicroelectronics/STM32CubeF4/issues/84

But for the F7 libs.

RCC_PLLCFGR_PLLSRC can't be compared with RCC_OscInitStruct->PLL.PLLSource. The PLLSource needs to be shifted over by the actual position in the register.

Same for PLLM.

RKOUSTM, 2021-09-03:
Hi @InconsolableCellist,

Thank you for your contribution. This request will be forwarded to our development teams for deeper analysis. We will be back to you as soon as they provide me with their answer. Thank you for you again for your contribution.

This duplicated issue will remain closed. Issue [#84](https://github.com/STMicroelectronics/STM32CubeF4/issues/84) that you reported will be kept open to track the point.

With regards,

RKOUSTM, 2021-09-03:
Duplicate of [STM32CubeF4#84](https://github.com/STMicroelectronics/STM32CubeF4/issues/84).

closed 2021-09-03 by RKOUSTM