STMicroelectronics/STM32CubeF7 #51: stm32f7xx_hal_uart.c:1557:48: error: 'HAL_DMA_ERROR_TIMEOUT' undeclared

fujiwara0, 2021-08-10:
**Describe the set-up**
 * STM32CubeF7 V1.16.1
 * gcc-arm-none-eabi-10.3-2021.07

**Describe the bug**
 * HAL_DMA_MODULE_ENABLED is disabled
 * Compilation of stm32f7xx_hal_uart.c fails
 * stm32f7xx_hal_uart.c is not checking the HAL_DMA_MODULE_ENABLED flag but is just using the DMA functions

`
Src\stm32f7xx_hal_uart.c
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c: In function 'HAL_UART_DMAStop':
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c:1557:48: error: 'HAL_DMA_ERROR_TIMEOUT' undeclared (first use in this function)
 1557 |         if (HAL_DMA_GetError(huart->hdmatx) == HAL_DMA_ERROR_TIMEOUT)
      |                                                ^~~~~~~~~~~~~~~~~~~~~
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c:1557:48: note: each undeclared identifier is reported only once for each function it appears in
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c: In function 'HAL_UART_Abort':
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c:1635:48: error: 'HAL_DMA_ERROR_TIMEOUT' undeclared (first use in this function)
 1635 |         if (HAL_DMA_GetError(huart->hdmatx) == HAL_DMA_ERROR_TIMEOUT)
      |                                                ^~~~~~~~~~~~~~~~~~~~~
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c: In function 'HAL_UART_AbortTransmit':
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c:1723:48: error: 'HAL_DMA_ERROR_TIMEOUT' undeclared (first use in this function)
 1723 |         if (HAL_DMA_GetError(huart->hdmatx) == HAL_DMA_ERROR_TIMEOUT)
      |                                                ^~~~~~~~~~~~~~~~~~~~~
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c: In function 'HAL_UART_AbortReceive':
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c:1782:48: error: 'HAL_DMA_ERROR_TIMEOUT' undeclared (first use in this function)
 1782 |         if (HAL_DMA_GetError(huart->hdmarx) == HAL_DMA_ERROR_TIMEOUT)
      |                                                ^~~~~~~~~~~~~~~~~~~~~
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c: In function 'HAL_UART_IRQHandler':
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c:2329:26: error: 'DMA_HandleTypeDef' {aka 'union __DMA_HandleTypeDef'} has no member named 'Init'
 2329 |         if (huart->hdmarx->Init.Mode != DMA_CIRCULAR)
      |                          ^~
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c:2329:41: error: 'DMA_CIRCULAR' undeclared (first use in this function)
 2329 |         if (huart->hdmarx->Init.Mode != DMA_CIRCULAR)
      |                                         ^~~~~~~~~~~~
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c: In function 'UART_DMATransmitCplt':
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c:3387:11: error: 'DMA_HandleTypeDef' {aka 'union __DMA_HandleTypeDef'} has no member named 'Init'
 3387 |   if (hdma->Init.Mode != DMA_CIRCULAR)
      |           ^~
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c:3387:26: error: 'DMA_CIRCULAR' undeclared (first use in this function)
 3387 |   if (hdma->Init.Mode != DMA_CIRCULAR)
      |                          ^~~~~~~~~~~~
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c: In function 'UART_DMAReceiveCplt':
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c:3439:11: error: 'DMA_HandleTypeDef' {aka 'union __DMA_HandleTypeDef'} has no member named 'Init'
 3439 |   if (hdma->Init.Mode != DMA_CIRCULAR)
      |           ^~
C://OS//compiler//ST//STM32CubeF7//Drivers//STM32F7xx_HAL_Driver//Src\stm32f7xx_hal_uart.c:3439:26: error: 'DMA_CIRCULAR' undeclared (first use in this function)
 3439 |   if (hdma->Init.Mode != DMA_CIRCULAR)
      |                          ^~~~~~~~~~~~
`

ASELSTM, 2021-08-11:
Hi @fujiwara0,

Indeed, the UART driver do not check on the **HAL_DMA_MODULE_ENABLED** state and will just use the functions of the DMA driver.

It's actually on the user application side that it should happen. The user will have to select the modules to activate in the same way as it is done in the configuration file **stm32f7xx_hal_conf.h** provided within all our examples and applications. As example you can check the [stm32f7xx_hal_conf.h](https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Projects/STM32F769I_EVAL/Examples/UART/UART_Printf/Inc/stm32f7xx_hal_conf.h) file of the UART_Printf example of the **STM32F769I_EVAL** board.

Users have to include the appropriate drivers that are necessary for their projects. Hence, you can use an **stm32f7xx_hal_conf.h** file and uncomment the needed modules or include manually the needed drivers and modules.

With regards,

ASELSTM, 2021-08-23:
Hi @fujiwara0,

Please allow me to close this thread. You can reopen it at any time if you have more details to provide us to help you.
Thank you for your comprehension and thank you once more for your contribution.

With regards,

closed 2021-08-23 by ASELSTM