STMicroelectronics/STM32CubeF7 #57: __LL_TIM_CALC_PSC float division rounding

phryniszak, 2021-11-01:
**Describe the set-up**
```
/**
  * @brief  HELPER macro calculating the prescaler value to achieve the required counter clock frequency.
  * @note ex: @ref __LL_TIM_CALC_PSC (80000000, 1000000);
  * @param  __TIMCLK__ timer input clock frequency (in Hz)
  * @param  __CNTCLK__ counter clock frequency (in Hz)
  * @retval Prescaler value  (between Min_Data=0 and Max_Data=65535)
  */
#define __LL_TIM_CALC_PSC(__TIMCLK__, __CNTCLK__)   \
  (((__TIMCLK__) >= (__CNTCLK__)) ? (uint32_t)(((__TIMCLK__)/(__CNTCLK__)) - 1U) : 0U)
```

Work as expected when the result of the division is integer, but with float compiler will always truncate.

So in my case when I call it with values:
`__LL_TIM_CALC_PSC(108*1e6, 10*1e6);`
function returns precaller value 9 which gives counter frequency `10.8*1e6 (8% error)`. The correct value is 10 with counter frequency equal `9.8*1e6 (2% error)`.

I didn't check but it looks like this macro is present in most of stm32cube packages in the same form.

**Additional context**
https://www.google.com/search?q=cpp+round+float+to+int

ASELSTM, 2021-11-29:
Hi @phryniszak,

Thank you for your report. Indeed, the issue is confirmed. To fix it you can use the patch below : 

```diff
/**
  * @brief  HELPER macro calculating the prescaler value to achieve the required counter clock frequency.
  * @note ex: @ref __LL_TIM_CALC_PSC (80000000, 1000000);
  * @param  __TIMCLK__ timer input clock frequency (in Hz)
  * @param  __CNTCLK__ counter clock frequency (in Hz)
  * @retval Prescaler value  (between Min_Data=0 and Max_Data=65535)
  */
#define __LL_TIM_CALC_PSC(__TIMCLK__, __CNTCLK__)   \
- (((__TIMCLK__) >= (__CNTCLK__)) ? (uint32_t)(((__TIMCLK__)/(__CNTCLK__)) - 1U) : 0U)
+ (((__TIMCLK__) >= (__CNTCLK__)) ? (uint32_t)((((__TIMCLK__) + (__CNTCLK__)/2U)/(__CNTCLK__)) - 1U) : 0U)
```
The fix will be made available in the frame of a future release.

With regards,

ASELSTM, 2021-11-29:
ST Internal Reference: 118234

ASELSTM, 2022-07-08:
Hi @phryniszak,

Thank you for your contribution. This issue has been fixed in the frame of version v1.17.0 of the STM32CubeF7. Please allow me then to close this thread.

Thank you again for your contribution.

With regards,

closed 2022-07-08 by ASELSTM