STMicroelectronics/STM32CubeF7 #59: Startup files set DMA2_Stream4_IRQHandler twice

higaski, 2021-12-28:
**Describe the bug**
The IRQ handler _DMA2_Stream4_IRQHandler_ is defined twice in every startup file inside the whole repository.
https://github.com/STMicroelectronics/STM32CubeF7/blob/f8cefdf02e8ad7fd06bd38a2dd85a55ecdbe92a9/Drivers/CMSIS/Device/ST/STM32F7xx/Source/Templates/gcc/startup_stm32f722xx.s#L466-L470

**How To Reproduce**
This issue only occurs when compiling any of the startup files with Clang. It produces the following error message.

`error: invalid reassignment of non-absolute variable 'DMA2_Stream4_IRQHandler'`

**Additional context**
From my understanding of the uses .thumb_set assembly directive this is actually a Clang bug and not necessarily an issue with the startup files. I'd still recommend to fix this though...

ASELSTM, 2022-01-06:
ST Internal Reference: 120364

ASELSTM, 2023-04-06:
Hi @higaski,

Thank you for your contribution. This issue has been fixed in the frame of version v1.17.0 of the STM32CubeF7. Please allow me then to close this thread.

With regards,

closed 2023-04-06 by ASELSTM