STMicroelectronics/STM32CubeF7 #60: Issue to call HAL_QSPI_Command_IT on RTOS, the interrupt continuous on QUADSPI_IRQHandler

pnt325, 2022-01-04:
**Description**
I try to development [QSPI_ReadWrite_IT](https://github.com/STMicroelectronics/STM32CubeF7/tree/master/Projects/STM32746G-Discovery/Examples/QSPI/QSPI_ReadWrite_IT) example on stm32f746G-disco. It's success run on bare-metal example. But when I try to run inside freeRTOS task. The interrupt callback continuous buf couldn't call to any Cmd/Rx/Tx Cplt method.

**Issue**
The problem at first time call: HAL_QSPI_Command_IT
```
if (HAL_QSPI_Command_IT(&hqspi, &sCommand) != HAL_OK) {
Error_Handler();
}
```
The interrupt callback continuous:
```
void QUADSPI_IRQHandler(void)
{
  /* USER CODE BEGIN QUADSPI_IRQn 0 */

  /* USER CODE END QUADSPI_IRQn 0 */
  HAL_QSPI_IRQHandler(&hqspi);
  /* USER CODE BEGIN QUADSPI_IRQn 1 */

  /* USER CODE END QUADSPI_IRQn 1 */
}
```
But wihtout jump to any callback  of :
```
/**
 * @brief  Rx Transfer completed callbacks.
 * @param  hqspi: QSPI handle
 * @retval None
 */
void HAL_QSPI_RxCpltCallback(QSPI_HandleTypeDef *hqspi) {
	RxCplt++;
}

/**
 * @brief  Tx Transfer completed callbacks.
 * @param  hqspi: QSPI handle
 * @retval None
 */
void HAL_QSPI_TxCpltCallback(QSPI_HandleTypeDef *hqspi) {
	TxCplt++;
}

/**
 * @brief  Status Match callbacks
 * @param  hqspi: QSPI handle
 * @retval None
 */
void HAL_QSPI_StatusMatchCallback(QSPI_HandleTypeDef *hqspi) {
	StatusMatch++;
}
```
It's alway  jump to: HAL_QSPI_FifoThresholdCallback the callback stack as bellow (debug on Ozone Segger): 
[image]

**Question**
1. Are there QSPI_HAL interrupt (DMA or IT) can run with RTOS task?
2. Any way to solve this issue?

Thank you!

RKOUSTM, 2022-01-07:
Hi @pnt325,

Thank you for your contribution. The point you reported is treated at [ST Community](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus) level. Please allow me to redirect you to address your request in [this](https://community.st.com/s/question/0D53W00000FPnh3/-stm32f7-quadspi-halqspicommandit-can-not-jump-to-quadspiirqhandler) discussion section where the same issue is addressed.  I hope the link I provided you with was helpful. I really hope you could find the root cause of this issue and, ideally, solve it.

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to a user application, please allow me to close it.

Thank you for your comprehension. We are looking forward to reading from you again.

With regards,

closed 2022-01-07 by RKOUSTM

pnt325, 2022-01-08:
Hi @RKOUSTM 

Thank for your response.
As your reference i already read. But it can not help  solve my problem. 

Finally i found where is problem but it's not clear. By the way change the QSPI interrupt priority lowest (to 15).