STMicroelectronics/STM32CubeF7 #62: PWR_STOP_RTC examplesdoesn't work

oscarhou, 2022-01-21:
**Describe the set-up**
STM32F769I_EVAL 
STM32CubeIDE 1.7.0

**Describe the bug**
PWR_STOP_RTC immediately exits STOP mode after entering.
https://github.com/STMicroelectronics/STM32CubeF7/tree/master/Projects/STM32F769I_EVAL/Examples/PWR/PWR_STOP_RTC

This issue is probably not restricted to the STM32F769I_EVAL example

**How To Reproduce**
1. In STM32CubeIDE use "Open Project From File System" to open the PWR_STOP_RTC example.
2. Build and run the code.
3. Observe that LED1 stays on instead of staying on for 5 seconds and turning off for ~20 seconds

**The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).**
Systick not being suspended before entering STOP mode.

**The use case that generates the problem.**
Running the example

**How we can reproduce the problem.**
Follow the reproduction steps above.

ALABSTM, 2022-08-04:
Hi @oscarhou,

We confirm the abnormal behavior you reported. It has been reported to our development teams for deeper analysis. We will get back to you as soon we have updates.

With regards,

ALABSTM, 2022-08-04:
ST Internal Reference: 132752

ALABSTM, 2022-10-18:
Hi @oscarhou,

I hope you are fine.  Back to you about this topic. Actually, the behaviour you are observing is **not** a bug. Below the details.

As indicated in the `readme.txt` file, this example **cannot** be run while debug mode is enabled.

https://github.com/STMicroelectronics/STM32CubeF7/blob/f8bda023e34ce9935cb4efb9d1c299860137b6f3/Projects/STM32F769I_EVAL/Examples/PWR/PWR_STOP_RTC/readme.txt#L55-L57

## Root cause of the observed behaviour
The behaviour you observed is due to the fact that you ran the example most likely **in debug mode**. Indeed, as explained in the reference manual [RM0385](https://www.st.com/resource/en/reference_manual/dm00124865-stm32f75xxx-and-stm32f74xxx-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf) _Rev.8_ pp. 1689-1690, for the debugger to keep connection to the core while in some low-power mode:

> [The] debugger host must first set some debug configuration registers to **change the low-power mode behavior**.
> In Stop mode, the **bit DBG_STOP must be previously set** by the debugger. This will enable the internal RC oscillator clock to feed FCLK and HCLK in STOP mode.

Said otherwise, while in debug mode, the core is **not allowed** to enter the stop mode.

## Running the example in _Release_ mode using CubeIDE

[image]

Source = [community.st.com](https://community.st.com/s/question/0D53W00001KzmHwSAJ/switching-from-debug-to-release-mode-how-do-i-push-firmware-to-target-beginner-question-am-i-doing-something-wrong).

## Possible solutions while in debug mode

* In case you **inadvertently** ran the example in debug mode, exit the debug mode simply by unplugging then plugging again the power supply cable of your board. Notice a simple push on the board's reset button is **not** sufficient.
* In case you would like to run the example in debug mode, you can force clearing the bit DBG_STOP by calling function `HAL_DBGMCU_DisableDBGStopMode()` just before entering the stop mode. 
https://github.com/STMicroelectronics/STM32CubeF7/blob/f8bda023e34ce9935cb4efb9d1c299860137b6f3/Drivers/STM32F7xx_HAL_Driver/Src/stm32f7xx_hal.c#L500-L503

## The Systick timer and the STOP mode

Regarding the solution you suggested, we tested it and observed indeed that the examples started running correctly after the Systick timer has been suspended. However, always according to the reference manual [RM0385](https://www.st.com/resource/en/reference_manual/dm00124865-stm32f75xxx-and-stm32f74xxx-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf) _Rev.8_ pp. 112, the Systick timer is **not** available in STOP mode.

[image]

Hence, although suspending the Systick timer **seemed** to cause the example to run correctly, it could be not adopted as the solution.

I hope my reply helped clarify this point and provided useful info. Do not hesitate if you have other questions related. Please allow me to close this thread.

With regards,

closed 2022-10-18 by ALABSTM