STMicroelectronics/STM32CubeF7 #63: ADC1, ADC2 no continuous conversion mode possible in STM32CubeMX when ADC1 enabled

rudi219, 2022-02-19:
**Describe the set-up**
 * STM32CubeMX V6.4
 * STM32CubeF7 Firmware V1.16.1 or 1.16.2

**Describe the bug**
It is not possible to enable continuous conversion mode for ADC2 or ADC3 when ADC1 is enabled.
It worked with Firmware V1.16.0 and before and is independent of the STM32CubeMX version (tested with V6.2, 6.3, 6.4)

**How To Reproduce**
1. Create a new project in STM32CubeMX for STM32F765Z
2. Enable ADC1 by enabling one channel
3. Enable ADC2 by enabling one channel
4. Enabling continuous conversion mode for ADC2 is not possible (same for ADC3)

ASELSTM, 2022-02-22:
Hi @rudi219,

Thank you for this report. The point you reported is actually related to the **CubeMX** generation problem and not to the firmware published in this repository. Unfortunately, we don't treat aspect related to CubeMX tool in our GitHub repositories.  They are rather treated at the [STM32CubeMX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated page of the **ST Community** . 

Since this issue is not directly related to the STM32Cube firmware but rather to our ecosystem, please allow me then to close this thread. Thank you for your comprehension. 

With regards,

closed 2022-02-22 by ASELSTM

rudi219, 2022-02-22:
Hello,are you sure, that it is not related to the firmware? It works with firmware up to 16.0 and different CubeMX versions, but not newer  firmware versions.Best regards,Rudi