STMicroelectronics/STM32CubeF7 #65: Nucleo F746ZG UART with DMA not working

xRadx, 2022-03-01:
**Describe the set-up**
 * Nucleo F746ZG
 * STM32CubeIDE 1.8.0 package 1.16.2

**Describe the bug**
UART RX/TX works in blocking mode, but after enabling DMA, the program does not call HAL_UART_RxCpltCallback.

**How To Reproduce**
Create project with nothing initialized. enable USART3 in async and activate DMA for TX/RX. Generate code. Create buffer and in User code 2 start UART receive with DMA (ie: HAL_UART_Receive_DMA(&huart3, buff, 4);

create PFP:
void HAL_UART_RxCpltCallback(UART_HandleTypeDef *huart) {
	HAL_GPIO_TogglePin(GPIOB, GPIO_PIN_0);
	HAL_UART_Receive_DMA(&huart3, buff, 4);
}

Upload code to the board.

After sending data >4 chars the LED should blink, but it does not.

ASELSTM, 2022-03-23:
Hi @xRadx,

Would you please share your project in order to allow a better analysis of the problem.

With regards,

xRadx, 2022-03-23:
Just .ioc and main.c would be sufficient?

xRadx, 2022-03-29:
after update of CubeMX and soft package the problem is gone. issue for closing

closed 2022-03-29 by xRadx