STMicroelectronics/STM32CubeF7 #66: How to generate fontX.c file?

MingYuanJeng, 2022-04-18:
How to generate font8.c、font16.、font24.c、、、fontX.c files?
Is there a tool to automatically generate it?
thanks

KamilDuljas, 2023-04-19:
Any information in this topic?

ALABSTM, 2023-08-07:
Hi @MingYuanJeng and @KamilDuljas,

Please excuse this delayed reply. Your question has been forwarded to our development teams. I'll get back to you as soon as I have their answer. Thank you again for your patience.

With regards,

ALABSTM, 2023-08-18:
Hi @MingYuanJeng, @KamilDuljas,

Here you are. The Fonts are generated using the “Dot Factory” tool. It is a small open source tool (**MIT licensed**) that can be downloaded from [here](https://github.com/pavius/the-dot-factory). A short tutorial and the revision history can also be found [here](http://www.eran.io/the-dot-factory-an-lcd-font-and-image-generator/).

I hope this answers your question. Please allow me to close this thread. Thank you for your patience.

With regards,

closed 2023-08-18 by ALABSTM