STMicroelectronics/STM32CubeF7 #73: Middleware/ST/TouchGFX, etc. links in  readme.txt don't work

jponko, 2022-09-01:
example:
https://www.st.com/en/licensed-software/touchgfx-mw.html?softwareProduct=TouchGFX-MW&softwareVersion=4.10.0

ASELSTM, 2022-09-14:
Hi @jponko,

Thank you for pointing this out. Indeed, we are experiencing some issues with these links. We recommend to download the full zipped package from [st.com](https://www.st.com/en/embedded-software/stm32cubeh7.html) as a workaround.

With regards,

TOUNSTM, 2023-04-10:
Hello @jponko,

As @ASELSTM said it's recommended to download the full zipped package from [st.com](https://www.st.com/en/embedded-software/stm32cubeh7.html). For more details, please refer to this [Link](https://github.com/STMicroelectronics/STM32CubeH7#some-middleware-libraries-are-unavailable-in-this-repository)

Please allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2023-04-10 by TOUNSTM