STMicroelectronics/STM32CubeF7 #74: USB FS Host MSC STM32F7 cube 1.17.0

FerdinandEn, 2022-09-17:
USB Host FS Driver for MSC does not work.

After upgrading one of my programs to the latest version of the Cube-HAL, my program with USB Host (USB FS) MSC on a STM32F746-Nucleo board stopped working. I use RTOS. The program still recognizes the device and then stops.

```
Application State: 0
USB Device Connected
USB Device Reset Completed
PID: 6544h
VID: 30deh
Address (#1) assigned.
Manufacturer : KIOXIA
Product : TransMemory
Serial Number : 0022CFF6B899C411F31022D1
Enumeration done.
This device has only 1 configuration.
Default configuration set.
Switching to Interface (#0)
Class    : 8h
SubClass : 6h
Protocol : 50h
MSC class started.
Number of supported LUN: 1
LUN #0:
```
Replacing the file `usbh_ctlreq.c` with version from 1.16.2 the program works again.

```
Application State: 0
USB Device Connected
USB Device Reset Completed
PID: 6544h
VID: 30deh
Address (#1) assigned.
Manufacturer : KIOXIA
Product : TransMemory
Serial Number : 0022CFF6B899C411F31022D1
Enumeration done.
This device has only 1 configuration.
Default configuration set.
Switching to Interface (#0)
Class    : 8h
SubClass : 6h
Protocol : 50h
MSC class started.
Number of supported LUN: 1
LUN #0:
Inquiry Vendor  : KIOXIA
Inquiry Product : TransMemory
Inquiry Version :
MSC Device ready
MSC Device capacity : 2615672320 Bytes
Block number : 30274559
Block Size   : 512
```
Also the demo project for the STM32746G-Discovery\Applications\USB_Host\MSC_RTOS has the same behavior as my program.

GLASS-FR, 2022-09-29:
Have a look to https://github.com/STMicroelectronics/stm32_mw_usb_host/issues/8...

FerdinandEn, 2022-09-30:
@GLASS-FR I have used the version 3.5.0. After that everything worked as before. Thanks for the tip

closed 2022-09-30 by FerdinandEn

GLASS-FR, 2022-10-01:
Hi, please don't close this issue. ALL HAL/CUBE PACKAGE MUST BE UPDATED . USB HOST V3.5.0 solve fatal issue of V3.4.1 MSC.
@ALABSTM : lot of users fall in this trap. I try to share existing solution in ST github
Why middelware repo are not crosslinked by all Hal using it?

ALABSTM, 2022-10-14:
Hi @GLASS-FR,

There is no plan for the moment to upgrade USB libraries to the latest versions within this firmware. As suggested by @FerdinandEn, you can find it within the [stm32_mw_usb_host](https://github.com/STMicroelectronics/stm32_mw_usb_host) repo.

With regards,

gcmitec, 2022-10-18:
Hi @GLASS-FR,
using F7 V1.17.0 from CubeMx try to set USBH_MAX_DATA_BUFFER to 1025.
Ciao.