STMicroelectronics/STM32CubeF7 #75: GUI_Delay STemWin doesn't work as expected for STM32769I-DISCO

xpress-embedo, 2022-09-17:
Hi All,
3 years ago, I faced a problem with the GUI_Delay function from STemWin as it was not working as expected.
I found similar issues all over the community pages; here are some links.
https://community.st.com/s/question/0D50X0000BMFJTpSQP/stemwin-guidispstring-function-isnt-working-after-a-while
https://community.st.com/s/question/0D53W000006pu0mSAA/im-working-on-stm32f469-discovery-board-for-gui-application-using-stm32cubemx-54-and-graphics-framework-stemwin-where-i-want-to-display-integers-from-1-to-5-my-delay-function-is-not-working-i-guess-only-1st-value-1-is-displaying-on-the-screen
https://community.st.com/s/question/0D53W00000wWPoXSAW/why-does-not-the-lcd-update-in-stm32h747idisco-board

And then in January 2022, I created a post on the community.
https://community.st.com/s/question/0D53W00001IcqibSAB/stemwinhelloworld-example-debugging-and-guidelay-not-working
https://community.st.com/s/question/0D53W00001IBcAxSAL/help-needed-in-building-a-simple-project-with-stemwin-and-stm32cubeide
But no solution was provided to me, to get better support I created a ticket on the "Online Support Case" page.
Case Number: 00149151
https://community.st.com/s/case/5003W00000EoYjjQAF/stemwin-graphical-framework-support-along-with-stm32f7-microcontroller

The following are the responses I have received.
[image]

[image]

So, technically they have acknowledged the problem, but no solution is provided till now.

Now coming to the problem, although it is mentioned in the community post.
I took the simple "STemWin_HelloWorld" example from the CubeF7 package for my board and updated it to print another string with some delay.
```
void MainTask(void) {
  GUI_Clear();
  GUI_SetFont(&GUI_Font32_1);
  GUI_DispString("Hello World from STM32F7\r\n");
  GUI_Delay(1000);
  GUI_DispStringAt("Hello world!", (LCD_GetXSize()-100)/2, (LCD_GetYSize()-20)/2);  
  while(1);
}
```
And this GUI_Delay is not working, I have STM32F4 DISCO Board also and also STM32F3 based Board and this works fine, but not on STM32769I-DISCO Board (maybe on older version this doesn't work, but on the newer version, it works but I don't have, and I am saying this because .. please see the comment from the post as attached below)

[image]

**Why I created this ticket?**
The main purpose of creating this ticket is to highlight this issue here so that developers can analyze and fix this issue because today it's almost 8 months since I didn't get support from the community post and also from the online support team.
Apologies if this is not the correct page to post this issue. Thanks

HBOSTM, 2023-02-17:
Hello @xpress-embedo,

Thank you your contribution. Actually, the point you raised has already been identified and is under study.

Best Regards,

xpress-embedo, 2024-06-14:
@TOUNSTM I checked today and the issue is still present and not fixed.