STMicroelectronics/STM32CubeF7 #77: max bytes limitation on function HAL_I2C_Mem_Read_DMA() using STM32F746IG and FW1.17.0

aalorencato, 2022-11-17:
**Describe the bug**

I'm using the STM32F746 core with HAL 1.17.0 to read bytes from the FUJITSU FRAM MB85RC256V on a custom board.

To read data through the I2C bus, I use the HAL_I2C_Mem_Read_DMA() function.

Up to HAL 1.16.2 I was able to read 512 bytes using just one function call. In HAL 1.17.0, if I try to read more than 256 bytes, the clock signal remains low forever.

I´m using STM32Cube 1.10.1 and default workspace setting (GNU Tools for STM32 10.3-2021.10)

**How To Reproduce**

My test code is:
```c
int main(void)
{
  /* USER CODE BEGIN 1 */

  /* USER CODE END 1 */

  /* MPU Configuration--------------------------------------------------------*/
  MPU_Config();

  /* Enable I-Cache---------------------------------------------------------*/
  SCB_EnableICache();

  /* Enable D-Cache---------------------------------------------------------*/
  SCB_EnableDCache();

  /* MCU Configuration--------------------------------------------------------*/

  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* USER CODE BEGIN Init */

  /* USER CODE END Init */

  /* Configure the system clock */
  SystemClock_Config();

  /* USER CODE BEGIN SysInit */

  /* USER CODE END SysInit */

  /* Initialize all configured peripherals */
  MX_GPIO_Init();
  MX_DMA_Init();
  MX_I2C1_Init();
  /* USER CODE BEGIN 2 */
  /* USER CODE END 2 */

  /* Infinite loop */
  /* USER CODE BEGIN WHILE */
  while (1)
  {
	  memset(_Buffer, 0, 512);
	  HAL_I2C_Mem_Read_DMA(&hi2c1, 0xAE, 0x000, I2C_MEMADD_SIZE_16BIT, _Buffer, 257);
	  HAL_Delay(1000);

    /* USER CODE END WHILE */

    /* USER CODE BEGIN 3 */
  }
  /* USER CODE END 3 */
}
```

1. Indicate the global behavior of your application project.
My test test code works well in HAL 1.16.2 for F7 family and in HAL 1.10.0 for H7 family.
Using HAL 1.17.0 for F7, the I2C stops after call HAL_I2C_Mem_Read_DMA with 257 or more bytes.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
I believe there was some bug in the implementation of the new interrupt handler (I2C_Mem_ISR_DMA) in version 1.17.0 .

3. The use case that generates the problem.
I believe that the STOP BIT is not being generated correctly inside the new interrupt handler I2C_Mem_ISR_DMA after the last byte.
look osciloscope images in screenshot section  (I´m sorry .. I dont have a logic analizer)

4. How we can reproduce the problem.
Just run the code above and change the number to read in function HAL_I2C_Mem_Read_DMA.
I tried using FRAM MB85RC256V but I understand that it can happen on any device that supports reading more than 256 bytes.

**ST Forum discussion**
[Is there any max bytes limitation on function HAL_I2C_Mem_Read_DMA() using STM32F746IG and FW1.17.0 ?](https://community.st.com/s/question/0D53W00001vdufzSAA/is-there-any-max-bytes-limitation-on-function-hali2cmemreaddma-using-stm32f746ig-and-fw1170-)

**Screenshots**
-> Communication OK - until 256 bytes in one read command (Blue is Clock .. it goes high after communication)

[image]

-> Communicatin FAIL - more than 256 bytes in one read command (Blue is Clock .. it goes low forever and I2C stops)

[image]

HBOSTM, 2022-12-16:
ST Internal Reference: 141554

HBOSTM, 2023-01-10:
Hello @aalorencato,

Thank you for this contribution, This issue raised as with the new version of I2C driver we fully manage all memory communication sequence through interrupt.
The problem is not camming from HAL implementation. To solve the issue It is recommended to have DMA higher priority than I2C, as the reload is on but through callback.
I share with you bellow an example of NVIC setting that you can use inside your project.

``` C
  // stm32f7xx_hal_msp.c

  HAL_NVIC_SetPriority(I2Cx_ER_IRQn, 1, 0);
  HAL_NVIC_EnableIRQ(I2Cx_ER_IRQn);
  HAL_NVIC_SetPriority(I2Cx_EV_IRQn, 2, 0);
  HAL_NVIC_EnableIRQ(I2Cx_EV_IRQn);
```
Could you please try with this modification and came back to me.

With regards,

ALABSTM, 2024-03-06:
Fixed in commit c7e2944ab9bcdea381e9326d1b90f9c4af59bd52

ALABSTM, 2024-03-06:
Hi @aalorencato,

I hope you are fine. Just to let you know this issue has been fixed and the fix just been uploaded. Thank you again for your detailed report.

With regards,

closed 2024-03-06 by ALABSTM