STMicroelectronics/STM32CubeF7 #78: SDMMC_DATATIMEOUT use in two different clock situations

elmagnificogi, 2022-12-01:
SDMMC_DATATIMEOUT use not right

I have check all thread about SDMMC_DATATIMEOUT

like:
> https://community.st.com/s/question/0D53W00000Toup2SAB/error-with-timeouts-constants-in-stm32f7xxhalsdc-in-sdsendsdstatus-and-sdfindscr-cause-indefinite-hangs
and issue # 53 #54 #55

There are two ways of use SDMMC_DATATIMEOUT 

One: use as cmd timeout or some status timeout, but not use HAL_GetTick use HaL clock, 1Mhz(typical)
```
  while ((HAL_SD_GetCardState(hsd) != HAL_SD_CARD_TRANSFER))
  {
    if ((HAL_GetTick() - tickstart) >=  SDMMC_DATATIMEOUT)
    {
      hsd->ErrorCode = HAL_SD_ERROR_TIMEOUT;
      hsd->State = HAL_SD_STATE_READY;
      return HAL_TIMEOUT;
    }
  }
```
Another: use as SDMMC_DTIMER register value that use SD clock  24Mhz(typical)
```
    /* Configure the SD DPSM (Data Path State Machine) */
    config.DataTimeOut   = SDMMC_DATATIMEOUT;
    config.DataLength    = BLOCKSIZE * NumberOfBlocks;
    config.DataBlockSize = SDMMC_DATABLOCK_SIZE_512B;
    config.TransferDir   = SDMMC_TRANSFER_DIR_TO_CARD;
    config.TransferMode  = SDMMC_TRANSFER_MODE_BLOCK;
    config.DPSM          = SDMMC_DPSM_DISABLE;
    (void)SDMMC_ConfigData(hsd->Instance, &config);
```

So it's different one, the timeout cant reach in real. 

I think you should differ the two ways and use two macro not one

I also check developer replays about consuming serval days to transfor one big file, it's not about the problem. The two value should different.

In fact , i have met the situation that SDMMC_DTIMER and wait status timeout contradict each other. If use too big SDMMC_DATATIMEOUT ,my program cant break from the timeout situation.If use too small SDMMC_DATATIMEOUT ,the sd cant read or write any more (could init)

HBOSTM, 2023-01-18:
Hello @elmagnificogi 

the use of SDMMC_DATATIMEOUT defined in the stm32f7xx_ll_sdmmc.h is conditioned by the non-definition of SDMMC_DATATIMEOUT by the user. In all cases, the DATATIMEOUT can be defined by user according to the application.
https://github.com/STMicroelectronics/STM32CubeF7/blob/bfaa85d92273fabe523724325f8d42001b497c62/Drivers/STM32F7xx_HAL_Driver/Inc/stm32f7xx_ll_sdmmc.h#L303 

This point is fixed in this [issue](https://github.com/STMicroelectronics/STM32CubeF7/issues/20).

With Regards,

elmagnificogi, 2023-01-18:
> 

> of

I know  user could dinfine , but you see my example code , it could use in two different clock , so it should use two macro not one do all things

HBOSTM, 2023-01-19:
ST Internal Reference: 143350

HBOSTM, 2023-01-19:
Hello @elmagnificogi ,

This point has been reported to our development teams for deeper analysis. We will get back to you as soon as we have updates.

With regards,

RJMSTM, 2024-06-20:
Fixed in : 043a9bf77465a9442127aac7307fdb36c9d47a36

closed 2024-06-20 by RJMSTM